openwrt-xiaomi/xmir-patcher #2: ERROR It is necessary to reorganize the device into "whc_cap" mode!

ArturWroblewski, 2022-11-09:
I have reset the router to factory settings.
I logged in automatically on http://router.miwifi.com/

I made the first configuration.

Administrator background: 192.168.31.1
Administrator password: 12345678

Run !START.bat and I got an error

==========================================================

Xiaomi MiR Patcher

 1 - Set IP-address (current value: 192.168.31.1)
 2 - Connect to device (install exploit)
 3 - Read full device info
 4 - Create full backup
 5 - Install EN/RU languages
 6 - Install Breed bootloader
 7 - Install firmware (from directory "firmware")
 8 - {{{ Other functions }}}
 9 - [[ Reboot device ]]
 0 - Exit

Select: 2

device_name = RB01
rom_version = 1.0.35 release
mac address = 70:85:c2:b1:55:cd
Enter device WEB password: 12345678

ERROR: It is necessary to reorganize the device into "whc_cap" mode!

==========================================================

Xiaomi MiR Patcher

 1 - Set IP-address (current value: 192.168.31.1)
 2 - Connect to device (install exploit)
 3 - Read full device info
 4 - Create full backup
 5 - Install EN/RU languages
 6 - Install Breed bootloader
 7 - Install firmware (from directory "firmware")
 8 - {{{ Other functions }}}
 9 - [[ Reboot device ]]
 0 - Exit

Select:

After checking on PC, it looks like this:

telnet 192.168.31.1
XiaoQiang login: root
Password: 12345678
Login incorrect

ssh 192.168.31.1
ssh: connect to host 192.168.31.1 port 22: Connection refused

vstkl, 2023-02-15:
Having the same issue, i tried UART method, but that didn't work because for some reason i didn't get the boot console... 
someone has (this solution)[https://forum.openwrt.org/t/add-openwrt-support-for-xiaomi-redmi-ax6000/125008/132] but that doesnt work for me and it doesn't write anything into the logs, will further investigate

chrisrer, 2023-05-20:
same problems any solution?

DaHype, 2023-08-28:
I have the same Problem

"ERROR: It is necessary to reorganize the device into "whc_cap" mode!"

With my AX3000
device_name = RA82
rom_version = 1.4.21 release

I want to change the Int Firmware 1.4.21 to the Chinese one 1.0.46

Any Solution?

Thanks

jackmeehan, 2023-08-28:
> > ERROR: Partition list is empty!
> 
> So the `dmesg` log contains a lot of unnecessary information. You need to restart your device.

I am getting the same "ERROR: Partition list is empty!" with my AX3000 v1.4.21 RA82. 

I have tried rebooting the router and trying again immediately after boot however I unfortunately continue to get the same error.

Is there anything else I can try?

jackmeehan, 2023-08-31:
> I am getting the same "ERROR: Partition list is empty!" with my AX3000 v1.4.21 RA82.
>
> I have tried rebooting the router and trying again immediately after boot however I unfortunately continue to [...]

@remittor thanks for fixing that! I get another error when trying to install firmware, I will start a new issue thread for that

closed 2023-10-05 by remittor