openwrt-xiaomi/xmir-patcher #4: Xiaomi WR30U (MB05) looking for help

hqvv, 2023-05-14:
Hi Remittor ,

Thank you for your great works on Redmi AX6000 for vulnerability exploit on openwrt forum.
We have same problem of getting ssh access of a new device from xiaomi model WR30U (MB05)

Could you please kindly help to look at the device, and give us some guidance of way to exploit.

I have uploaded some resource files to my server hope it helps.
rootfs file https://hquu.net/r/rom/img-857011981_vol-rootfs.ubifs
lua file https://hquu.net/r/rom/wr30u_misystem.lua

You may find related post on forum https://forum.openwrt.org/t/openwrt-support-for-xiaomi-ax3000ne/153769
Thanks!

hqvv, 2023-05-17:
Closed as the problem was solved by  https://github.com/PatriciaLee3/wr30u_ssh 

Thanks

closed 2023-05-17 by hqvv

remittor, 2023-10-19:
This device supported!