openwrt-xiaomi/xmir-patcher #5: Issue installing firmware on RA82 AX3000 INT

jackmeehan, 2023-08-31:
Thanks to @remittor for fixing the issue with reading device info from the RA82 so quickly!

The exploit now runs, the script can retrieve the device info, however I'm still having trouble installing firmware.

My router is currently on stock v1.4.21 and I am trying to downgrade to stock v1.3.27 (rx power issues on 1.4.21). I get the following output from the tool:

> Detect valid SSH server on port 22 (auth OK)
> Image files in directory "firmware/":
>   "firmware/factory.bin"
> Download file: "/tmp/dmesg.log" ....
> Download file: "/tmp/mtd_list.txt" ....
> Download file: "/tmp/kcmdline.log" ....
>  
> ERROR: Kernel data not found!

I have tried installing v1.4.21 also and get the same error. I'm wondering if there is anything extra I need to do in order to successfully flash the RA82 or if it's just not possible?

Thanks in advance.

J

remittor, 2023-08-31:
I have looked at the contents of the image you are trying to flash. I have not done support for installing such images yet. I don't have a device that requires such images.

jackmeehan, 2023-08-31:
> I have looked at the contents of the image you are trying to flash. I have not done support for installing such images yet. I don't have a device that requires such images.

Thanks v much for replying and looking into it. With some modifications (with the help of Code Interpreter) I have gotten it as far as the RootFS check but got stuck there. 

Are you aware of any other tools that might help downgrade the stock firmware? Firmware Mod Kit and Binwalk don’t recognise the file system

remittor, 2023-09-02:
fixed: https://github.com/openwrt-xiaomi/xmir-patcher/commit/22f755be25ce51b8b8ecf02a553c9b0cc9a311f9

jackmeehan, 2023-09-03:
> fixed: [22f755b](https://github.com/openwrt-xiaomi/xmir-patcher/commit/22f755be25ce51b8b8ecf02a553c9b0cc9a311f9)

Thanks v much for updating. (FYI: there's a typo on line 205 of gateway.py ["foarmat"])

Unfortunately the tool still doesn't recognise the file system of stock firmware (I tried v1.3.27 and 1.4.21) - resulting in:

> Detect valid SSH server on port 22 (auth OK)
> Image files in directory "firmware/":
>   "firmware/miwifi_ra82_all_7eb36_1.3.27_INT.bin"
> Download file: "/tmp/dmesg.log" ....
> Download file: "/tmp/mtd_list.txt" ....
> Download file: "/tmp/kcmdline.log" ....
>  
> ERROR: Kernel data not found!

1.3.27 is the version I'm trying to install. I have attached both 1.3.27 and 1.4.21 firmware files here in case you don't have access to them. I've also attached the "output" dir in case there is anything useful to help debug.

Thanks again!

[outdir.zip](https://github.com/openwrt-xiaomi/xmir-patcher/files/12507018/outdir.zip)
[firmware](https://drive.google.com/drive/folders/1cKtdHwnpF08IZeaUjHVncSSpuUVg-3zc?usp=sharing)

remittor, 2023-10-19:
@jackmeehan , try with new version

ScotchTartan, 2023-10-20:
I'm using the same router and firmware than @jackmeehan. I cloned your last repo.

1- OK
2- OK
3- OK
4- OK
5- OK
6- No error but i didn't got any output. is it ok?
7- Problems ........

```
Detect valid SSH server on port 22 (auth OK)
device: "RA82"
img_write = True
Image files in directory "firmware/":
  "firmware/miwifi_ra82_all_7eb36_1.3.27_INT.bin"
Download file: "/tmp/dmesg.log" ....
Download file: "/tmp/mtd_list.txt" ....
Download file: "/tmp/mtd_addr.txt" ....
Download file: "/tmp/kcmdline.log" ....
Parse all images...
Stock image list: {'xiaoqiang_version': 537, 'uboot.bin': 507904, 'root.ubi': 25952256}
Parse HDR image for "RA82" router
UBI: filetype: b'UBI#'
UBI_File Warning: end_offset - start_offset length is not block aligned, could mean missing data.
UBI: Decoding UBIFS...
UBI:   volume: "kernel" 	 size: 3555328 
UBI:   volume: "ubi_rootfs" 	 size: 21331968 
Traceback (most recent call last):
```

  ```
File "/home/stallman/xmir-patcher-main/install_fw.py", line 1087, in <module>
    xf.parse_all_images()
  File "/homestallman/xmir-patcher-main/install_fw.py", line 174, in parse_all_images
    self.parse_stock_image(image)
  File "/home/stallman/xmir-patcher-main/install_fw.py", line 292, in parse_stock_image
    hr = self.parse_image(img.data, img.name)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/stallman/xmir-patcher-main/install_fw.py", line 212, in parse_image
    ubivol = self.parse_ubifs(image)
             ^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/osito/stallman/install_fw.py", line 630, in parse_ubifs
    ufile_obj.close()
    ^^^^^^^^^^^^^^^
AttributeError: 'ubi_file' object has no attribute 'close'

```
`

remittor, 2023-10-21:
@ScotchTartan , Your source code is not the latest

ScotchTartan, 2023-10-21:
> @ScotchTartan , Your source code is not the latest

I tried using latest firmware (miwifi_ra82_firmware_722ce_1.4.21_INT.bin) and previus (miwifi_ra82_all_7eb36_1.3.27_INT.bin)
I cloned the latest git repository, i'm sure because latest commits are included.

What do you mean with source code?

remittor, 2023-10-21:
@ScotchTartan , comment line with code `ufile_obj.close()`

ScotchTartan, 2023-10-21:
> @ScotchTartan , comment line with code `ufile_obj.close()`

```
Detect valid SSH server on port 22 (auth OK)
device: "RA82"
img_write = True
Image files in directory "firmware/":
  "firmware/miwifi_ra82_all_7eb36_1.3.27_INT.bin"
Download file: "/tmp/dmesg.log" ....
Download file: "/tmp/mtd_list.txt" ....
Download file: "/tmp/mtd_addr.txt" ....
Download file: "/tmp/kcmdline.log" ....
Parse all images...
Stock image list: {'xiaoqiang_version': 537, 'uboot.bin': 507904, 'root.ubi': 25952256}
Parse HDR image for "RA82" router
UBI: filetype: b'UBI#'
UBI_File Warning: end_offset - start_offset length is not block aligned, could mean missing data.
UBI: Decoding UBIFS...
UBI:   volume: "kernel" 	 size: 3555328 
UBI:   volume: "ubi_rootfs" 	 size: 21331968 
parse_ubifs = 2
 
ERROR: FIT: Incorrect image (0)
```

Sorry, i have been commented that line because the output error but forget to post the output

remittor, 2023-10-22:
@ScotchTartan , fixed https://github.com/openwrt-xiaomi/xmir-patcher/commit/e8b0a9fefe2906a7141f96b39b14dec19bd87ce0

ScotchTartan, 2023-10-22:
```
Detect valid SSH server on port 22 (auth OK)
device: "RA82"
img_write = True
Image files in directory "firmware/":
  "firmware/miwifi_ra82_all_7eb36_1.3.27_INT.bin"
Download file: "/tmp/dmesg.log" ....
Download file: "/tmp/mtd_list.txt" ....
Download file: "/tmp/mtd_addr.txt" ....
Download file: "/tmp/kcmdline.log" ....
Parse all images...
Stock image list: {'xiaoqiang_version': 537, 'uboot.bin': 507904, 'root.ubi': 25952256}
Parse HDR image for "RA82" router
UBI: filetype: b'UBI#'
UBI_File Warning: end_offset - start_offset length is not block aligned, could mean missing data.
UBI: Decoding UBIFS...
UBI:   volume: "kernel" 	 size: 3555328 
UBI:   volume: "ubi_rootfs" 	 size: 21331968 
parse_ubifs = 2
FIT size = 0x34FFC0 (3391 KiB)
FIT: name = "ARM OpenWrt FIT (Flattened Image Tree)"
FIT: def_cfg: "config@tb-mp04"
FIT: def_fdt: "fdt@tb-mp04"
FDT: desc = "ARM OpenWrt qcom-ipq50xx-mpxx device tree blob"
FDT: type = "flat_dt"
FDT: arch = "arm"
KRN: desc = "ARM OpenWrt Linux-4.4.60"
KRN: type = "kernel"
KRN: arch = "arm"
KRN: compression = "lzma"
KRN: data = 2040626 bytes
Traceback (most recent call last):
  File "/home/stallman/xmir-patcher/install_fw.py", line 1100, in <module>
    xf.parse_all_images()
  File "/home/stallman/xmir-patcher/install_fw.py", line 174, in parse_all_images
    self.parse_stock_image(image)
  File "/home/stallman/xmir-patcher/install_fw.py", line 292, in parse_stock_image
    hr = self.parse_image(img.data, img.name)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/stallman/xmir-patcher/install_fw.py", line 217, in parse_image
    kk = self.parse_fit(ubivol['kernel'], footer = False)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/stallman/xmir-patcher/install_fw.py", line 514, in parse_fit
    dt = fdt.parse_dtb(krn_dt_data)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/stallman/xmir-patcher/venv/lib/python3.11/site-packages/fdt/__init__.py", line 530, in parse_dtb
    assert isinstance(data, (bytes, bytearray)), "Invalid argument type"
AssertionError: Invalid argument type

```

ScotchTartan, 2023-10-22:
```
Select: 6

Detect valid SSH server on port 22 (auth OK)
 
ERROR: Breed bootloader cannot be installed on this device!
```
Seems this device has the bootloader locked?

remittor, 2023-10-22:
@ScotchTartan , fixed https://github.com/openwrt-xiaomi/xmir-patcher/commit/b1e2c7bf433e44b70005c56dd6b9f92b1b1329b8

ScotchTartan, 2023-10-22:
It works, flashed OK.

Thx

jackmeehan, 2023-10-28:
@remittor thanks for updates - stock firmware files now flash successfully to RA82! 

However, I still get the below error when trying to install Breed:

> 6
> 
> Detect valid SSH server on port 22 (auth OK)
>  
> ERROR: Breed bootloader cannot be installed on this device!

@ScotchTartan did you manage to get Breed installed successfully? If so, how?

Thanks

ScotchTartan, 2023-10-28:
> @ScotchTartan did you manage to get Breed installed successfully? If so, how?
> 
> Thanks

No, as far as i know, the bootloader (uboot) of this device is locked.
By the way, last stable image for RA82 is 1.4.31, but nobody dumped it.

When i flashed it, i can install any stock image version with permanent ssh access, but the advanced configuration is still blocked because the accessed GUI is the same. In the video published on YouTube you can access the original Luci gui with all the available configurations, unfortunately with ra82 i didn't. It's fine for downgrade firmware, people reported performance drops with latest  firmware. I'm not a master of OpenWrt.

closed 2023-11-03 by remittor