openwrt-xiaomi/xmir-patcher #8: ERROR: Partition list is empty!

NordGeit, 2023-10-07:
Trying to push in OpenWRT on this Xiaomi AX3200 RB01, which on the OpenWRT forum it apparently is meant to support.
I did the instructions, but when I try to install the firmware, I get this:
```
Image files in directory "firmware/":
  "firmware/factory.bin"
Download file: "/tmp/dmesg.log" ....

ERROR: Partition list is empty!
```
(the firmware is the correct one, I just renamed it)
Notably, option 3 and 4 give the same result (read full device info and create full backup respectively)

remittor, 2023-10-09:
I fixed this problem on August 30:
https://github.com/openwrt-xiaomi/xmir-patcher/commit/5deffbd1faf116e918b8d5bc52382cf8cd59076d

NordGeit, 2023-10-10:
> I fixed this problem on August 30: [5deffbd](https://github.com/openwrt-xiaomi/xmir-patcher/commit/5deffbd1faf116e918b8d5bc52382cf8cd59076d)

Any clue what could be causing this error, or an idea to let us get closer to the root cause?
I haven't updated the firmware, and I tried to log into telnet and it seems fine, getting the ARE U OK? ascii art prompt, however the router does just not want me to SSH into it.
I must add that this is my first time really fiddling with a router, so I don't have any fancy UART fiddling devices or a second router which there are apparently clear tutorials for.
Netmode is 4, and that is the only hope I have that doing this with nothing but my puters and single router is possible.

9mb-Project, 2023-10-18:
I encountered the following error while attempting to flush firmware for my Xiaomi ax3000

Select: 7
img_write = True
ERROR:` The firmware was not found in the "firmware/" folder!

May I ask what caused the problem

remittor, 2023-10-19:
@NordGeit , try with new version

ScotchTartan, 2023-10-22:
> I encountered the following error while attempting to flush firmware for my Xiaomi ax3000
> 
> Select: 7 img_write = True ERROR:` The firmware was not found in the "firmware/" folder!
> 
> May I ask what caused the problem

Did you put a firmware inside firnware folder?
Ir order to flash, first you need to download the desired firmware, then copy it inside ./firmware folder.

gbenamy, 2024-01-15:
Hi,
I've downloaded the latest code yesterday (to a raspberry pi2 to utilize)
enabling ssh works perfect, 
When trying to flash the firmware, I'm getting the following error:
`Select: 7

Detect valid SSH server on port 22 (auth OK)
device: "RB01"
img_write = True
Image files in directory "firmware/":
  "firmware/openwrt-23.05.2-mediatek-mt7622-xiaomi_redmi-router-ax6s-squashfs-factory.bin"
Download file: "/tmp/dmesg.log" ....
ERROR on downloading "/tmp/dmesg.log"
Download file: "/tmp/mtd_list.txt" ....

ERROR: Partition list is empty! (solution: disable all WiFi modules and reboot device)
`
(Xiaomi 3200ax, stock firmware 1.0.83)
I've tried to set 'off' to wifi via the router config page, still the same..
Is that what you meant by disable all WIFI?
Or to ssh to the router and start 'rmmod' wifi modules? 
Thanks

NordGeit, 2024-01-20:
Alright, sorry for the... 3 month delay, I got busy procrastinating.
I decided to run the newest version, same model and everything, after a reset on the router itself, which didn't work entirely, just re-did the wifi names and passwords...
Anyways, glad my phone remembered the admin for me, but now I get this when trying to select option 2:

`Enter device WEB password: cuckcuck
Traceback (most recent call last):
  File "connect.py", line 59, in <module>
    import connect5
  File "C:\Users\%user%\Downloads\xmir-patcher-main\xmir-patcher-main\connect5.py", line 264, in <module>
    res = get_all_scenes()
  File "C:\Users\%user%\Downloads\xmir-patcher-main\xmir-patcher-main\connect5.py", line 100, in get_all_scenes
    return exec_smart_command("get_scene_setting")
  File "C:\Users\%user%\Downloads\xmir-patcher-main\xmir-patcher-main\connect5.py", line 93, in exec_smart_command
    raise ExploitError(f'Error on exec command "{cmd}" => {res}')
connect5.ExploitError: Error on exec command "get_scene_setting" => {"code":-100,"msg":"connect failed"}`

Same old, still Windows 10.
This... Probably should be a _new_ issue, but I digress.

remittor, 2024-01-21:
> connect5.ExploitError: Error on exec command "get_scene_setting" => {"code":-100,"msg":"connect failed"}

This is due to a bug in the `smartcontroller` service. It’s probably worth resetting the entire firmware to the default state.