openwrt-xiaomi/xmir-patcher #9: Error creating full backup

kilroy98, 2023-10-29:
Router: Redmi AC2100
Stock firmware version: 2.0.23
```
Select: 4

Download file: "/tmp/dmesg.log" ....
Download file: "/tmp/mtd_list.txt" ....
Download file: "/tmp/mtd_addr.txt" ....
Download file: "/tmp/kcmdline.log" ....
Download file: "/tmp/dmesg.log" ....
Download file: "/tmp/mtd_list.txt" ....
Download file: "/tmp/mtd_addr.txt" ....
Full backup creating...

ERROR: SSH execute command timed out! CMD: "dd if=/dev/mtd0 of=/tmp/mtd_dump.bin"
```
After this error web get "502 Bad Gateway" and tool no more work. And need to reboot router for fix it.

remittor, 2023-10-30:
@kilroy98 
In this case, use options 8 - 4

kilroy98, 2023-10-30:
I tried this and backup was made only from 1 to 9. From 10 to 13 it fails with an error (like point 0)

```
Choice: 4

Download file: "/tmp/dmesg.log" ....
Download file: "/tmp/mtd_list.txt" ....
Download file: "/tmp/mtd_addr.txt" ....
Download file: "/tmp/kcmdline.log" ....
Download file: "/tmp/dmesg.log" ....
Download file: "/tmp/mtd_list.txt" ....
Download file: "/tmp/mtd_addr.txt" ....
   0 > addr: 0x00000000  size: 0x07F80000  name: "ALL"
   1 > addr: 0x00000000  size: 0x00080000  name: "Bootloader"
   2 > addr: 0x00080000  size: 0x00040000  name: "Config"
   3 > addr: 0x000C0000  size: 0x00040000  name: "Bdata"
   4 > addr: 0x00100000  size: 0x00040000  name: "Factory"
   5 > addr: 0x00140000  size: 0x00040000  name: "crash"
   6 > addr: 0x00180000  size: 0x00040000  name: "crash_syslog"
   7 > addr: 0x001C0000  size: 0x00040000  name: "cfg_bak"
   8 > addr: 0x00200000  size: 0x00400000  name: "kernel0"
   9 > addr: 0x00600000  size: 0x00400000  name: "kernel1"
  10 > addr: 0x00A00000  size: 0x01A00000  name: "rootfs0"
  11 > addr: 0x02400000  size: 0x01A00000  name: "rootfs1"
  12 > addr: 0x03E00000  size: 0x02600000  name: "overlay"
  13 > addr: 0x06400000  size: 0x01B80000  name: "obr"

Enter partition name or mtd number: 12

ERROR: SSH execute command timed out! CMD: "dd if=/dev/mtd12 of=/tmp/mtd_dump.bin"
```

But if you select items from 10 to 13, then there will be no error 502. Only if you making a full backup or select item 0.

kilroy98, 2023-11-01:
Is it normal for stock Xiaomi firmware that some partitions cannot be backed up or the backup function is not optimized for my router?

remittor, 2023-11-01:
It’s just that when writing the code, I didn’t take into account the fact that some devices have very small RAM.
For such cases, I need to rewrite my code.

kilroy98, 2023-11-01:
I understand. This router has 128MB of RAM.
If you plan to change the code in the near future, then I may not install custom firmware for now to check the functionality later.

kilroy98, 2023-11-05:
@remittor  How can I find out the size of the partition? I connected to the router via winscp, when I tried to create a backup from 10 to 13, the file was still created, even though the utility had an error message. But I don’t know what size these partitions are to be sure that its backup of this partition is full.

_P.S. And can I write here in Russian? Or is it better to use English?_

ge38kun, 2024-06-07:
I have made a full backup of Xiaomi 6500 pro. How can be restored? is there any option?

thanks

kilroy98, 2024-07-04:
I bought a Redmi AX6000, I’m trying to make a full backup and it’s not being created either, although there is more RAM.
```
Select: 4

Download file: "/tmp/dmesg.log" ....
Download file: "/tmp/mtd_list.txt" ....
Download file: "/tmp/mtd_info.txt" ....
Download file: "/tmp/mtd_fdt.txt" ....
Download file: "/tmp/kcmdline.log" ....
Download file: "/tmp/dmesg.log" ....
Download file: "/tmp/mtd_list.txt" ....
Download file: "/tmp/mtd_info.txt" ....
Download file: "/tmp/mtd_fdt.txt" ....
Full backup creating...

ERROR: SSH execute command timed out! CMD: "dd if=/dev/mtd0 of=/tmp/mtd_dump.bin"
```
I'm trying to make a backup of each partition separately - in the end, backups were made from partitions 1 to 7. Partition 0, 8, 9 and 10 failed.

```
Choice: 4

Download file: "/tmp/dmesg.log" ....
Download file: "/tmp/mtd_list.txt" ....
Download file: "/tmp/mtd_info.txt" ....
Download file: "/tmp/mtd_fdt.txt" ....
Download file: "/tmp/kcmdline.log" ....
Download file: "/tmp/dmesg.log" ....
Download file: "/tmp/mtd_list.txt" ....
Download file: "/tmp/mtd_info.txt" ....
Download file: "/tmp/mtd_fdt.txt" ....
   0 > addr: 0x00000000  size: 0x08000000  name: "spi0.1"
   1 > addr: 0x00000000  size: 0x00100000  name: "BL2"
   2 > addr: 0x00100000  size: 0x00040000  name: "Nvram"
   3 > addr: 0x00140000  size: 0x00040000  name: "Bdata"
   4 > addr: 0x00180000  size: 0x00200000  name: "Factory"
   5 > addr: 0x00380000  size: 0x00200000  name: "FIP"
   6 > addr: 0x00580000  size: 0x00040000  name: "crash"
   7 > addr: 0x005C0000  size: 0x00040000  name: "crash_log"
   8 > addr: 0x00600000  size: 0x01E00000  name: "ubi"
   9 > addr: 0x02400000  size: 0x01E00000  name: "ubi1"
  10 > addr: 0x04200000  size: 0x03200000  name: "overlay"

Enter partition name or mtd number: 0

ERROR: SSH execute command timed out! CMD: "dd if=/dev/mtd0 of=/tmp/mtd_dump.bin"
```

The only difference from the Redmi AC2100 is that the router does not freeze after such an error.

atolismesh, 2024-07-04:
@kilroy98 
Try change line 71 in create_backup.py from:
`  gw.run_cmd("dd if=/dev/mtd{id} of={o}".format(id=pid, o=fn_remote))`
to
`  gw.run_cmd("dd if=/dev/mtd{id} of={o}".format(id=pid, o=fn_remote), timeout=20)`
- set timeout to 20 sec or more

kilroy98, 2024-07-05:
> @kilroy98 Try change line 71 in create_backup.py from: ` gw.run_cmd("dd if=/dev/mtd{id} of={o}".format(id=pid, o=fn_remote))` to ` gw.run_cmd("dd if=/dev/mtd{id} of={o}".format(id=pid, o=fn_remote), timeout=60)` set timeout to 60 sec or more and line 99 change to ` ret = gw.run_cmd(cmd, timeout=60, die_on_error = False)`

Thanks, this helped. I tested on Redmi AX6000, but I no longer have the opportunity to test Redmi AC2100.

ortegafernando, 2024-08-12:
Not working in AX3000T RD03. [https://github.com/openwrt-xiaomi/xmir-patcher/issues/15](https://github.com/openwrt-xiaomi/xmir-patcher/issues/15)

ortegafernando, 2024-08-13:
Finally with wiki commands I get managed to flash openwrt. Thanks a lot !!!

closed 2025-10-19 by remittor