openwrt-xiaomi/xmir-patcher #12: Partition list is empty!

NecronomiconCoding, 2023-11-02:
Hi, 
I have a RA70 with 3.0.48 release. 

SSH is installed and running. 

When running 3 or 7 i get the Error Partition list is empty! 

Happens when its downloading dmesg.log (ERROR on downloading dmesg) and downloading mtd_list.txt

NecronomiconCoding, 2023-11-02:
If this would help you, 

`root@XiaoQiang:~# cat /proc/mtd
dev:    size   erasesize  name
mtd0: 00100000 00020000 "0:SBL1"
mtd1: 00100000 00020000 "0:MIBIB"
mtd2: 00080000 00020000 "0:BOOTCONFIG"
mtd3: 00080000 00020000 "0:BOOTCONFIG1"
mtd4: 00300000 00020000 "0:QSEE"
mtd5: 00300000 00020000 "0:QSEE_1"
mtd6: 00080000 00020000 "0:DEVCFG"
mtd7: 00080000 00020000 "0:DEVCFG_1"
mtd8: 00080000 00020000 "0:APDP"
mtd9: 00080000 00020000 "0:APDP_1"
mtd10: 00080000 00020000 "0:RPM"
mtd11: 00080000 00020000 "0:RPM_1"
mtd12: 00080000 00020000 "0:CDT"
mtd13: 00080000 00020000 "0:CDT_1"
mtd14: 00080000 00020000 "0:APPSBLENV"
mtd15: 00100000 00020000 "0:APPSBL"
mtd16: 00100000 00020000 "0:APPSBL_1"
mtd17: 00080000 00020000 "0:ART"
mtd18: 00080000 00020000 "bdata"
mtd19: 00080000 00020000 "crash"
mtd20: 00080000 00020000 "crash_syslog"
mtd21: 03800000 00020000 "rootfs"
mtd22: 03800000 00020000 "rootfs_1"
mtd23: 00100000 00020000 "cfg_bak"
mtd24: 07d80000 00020000 "overlay"
mtd25: 005ef000 0001f000 "kernel"
mtd26: 02093000 0001f000 "ubi_rootfs"
mtd27: 00a6a000 0001f000 "rootfs_data"
mtd28: 03013000 0001f000 "data"`

NecronomiconCoding, 2023-11-02:
I dont get that error anymore after your latest commit but the firmware installation isnt working: 

==========================================================

Xiaomi MiR Patcher

 1 - Set IP-address (current value: 192.168.31.1)
 2 - Connect to device (install exploit)
 3 - Read full device info
 4 - Create full backup
 5 - Install EN/RU languages
 6 - Install Breed bootloader
 7 - Install firmware (from directory "firmware")
 8 - {{{ Other functions }}}
 9 - [[ Reboot device ]]
 0 - Exit

Select: 2

device_name = RA70
rom_version = 3.0.48 release
mac address = a8:5e:45:58:0c:95
Telnet server already running, but FTP server not respond
Enter device WEB password: Noor9988
Enable smartcontroller scene executor ...
Wait smartcontroller activation ...
___[504]___
___[504]___
Unlock dropbear service ...
Unlock SSH server ...
Set password "root" for root user ...
Enabling dropbear service ...
Run SSH server on port 22 ...
Test SSH connection to port 22 ...

#### SSH server are activated! ####

==========================================================

Xiaomi MiR Patcher

 1 - Set IP-address (current value: 192.168.31.1)
 2 - Connect to device (install exploit)
 3 - Read full device info
 4 - Create full backup
 5 - Install EN/RU languages
 6 - Install Breed bootloader
 7 - Install firmware (from directory "firmware")
 8 - {{{ Other functions }}}
 9 - [[ Reboot device ]]
 0 - Exit

Select: 7

device: "RA70"
img_write = True
Image files in directory "firmware/":
  "firmware/openwrt-qualcommax-ipq807x-xiaomi_ax9000-initramfs-factory.ubi"
Download file: "/tmp/dmesg.log" ....
Download file: "/tmp/mtd_list.txt" ....
Download file: "/tmp/mtd_addr.txt" ....
Download file: "/tmp/kcmdline.log" ....
Parse all images...
UBI: filetype: b'UBI#'
UBI: Decoding UBIFS...
UBI:   volume: "kernel"          size: 12697600
parse_ubifs = 1
FIT size = 0xC117D4 (12357 KiB)
FIT: name = "ARM64 OpenWrt FIT (Flattened Image Tree)"
FIT: def_cfg: "config@hk14"
FIT: def_fdt: "fdt-1"
FDT: desc = "ARM64 OpenWrt xiaomi_ax9000 device tree blob"
FDT: type = "flat_dt"
FDT: arch = "arm64"
KRN: desc = "ARM64 OpenWrt Linux-6.1.60"
KRN: type = "kernel"
KRN: arch = "arm64"
KRN: compression = "gzip"
KRN: data = 12606614 bytes
FDT: compatible = ['xiaomi,ax9000', 'qcom,ipq8074']
FDT: model = "Xiaomi AX9000"
FDT: dt_part: ['/soc/nand-controller@79b0000/nand@0/partitions']
FIT: detect initrd into kernel image
fw_img: 13056 KiB | kernel: 12357 KiB | rootfs: 1 KiB
Download file: "/tmp/bl_0SBL1.bin" ....
Download file: "/tmp/bl_0APPSBL.bin" ....
Download file: "/tmp/env_0APPSBLENV.bin" ....
Download file: "/tmp/env_bdata.bin" ....
Download file: "/tmp/env_0SBL1.bin" ....
current flag_boot_rootfs = 0
install_method = 200
--------- prepare command lines -----------
fw_img: 13056 KiB | kernel: 12357 KiB | rootfs: 1 KiB
------------- flash images -------------
Upload file: "tmp/fw/fw_img.bin" ....
Run scripts for change NVRAM params...
Boot from firmware [1] activated.
Writing firmware image to addr 0x04980000 ...
  mtd -e "rootfs_1" write "/tmp/fw_img.bin" "rootfs_1"
The firmware has been successfully flashed!
Send command "reboot" via SSH/Telnet ...

ERROR: SSH execute command timed out! CMD: "reboot -f"

==========================================================

Device now reboots and its still the same firmware as before.

dimfishr, 2023-11-03:
You should check via SSH
```
nvram get flag_try_sys1_failed
nvram get flag_try_sys2_failed
```
If values are not 0 - reset them:

```
nvram set flag_try_sys1_failed=0
nvram set flag_try_sys2_failed=0
nvram commit
```

NecronomiconCoding, 2023-11-03:
> You should check via SSH
> 
> ```
> nvram get flag_try_sys1_failed
> nvram get flag_try_sys2_failed
> ```
> 
> If values are not 0 - reset them:
> 
> ```
> nvram set flag_try_sys1_failed=0
> nvram set flag_try_sys2_failed=0
> nvram commit
> ```

Both are 0 

`root@XiaoQiang:~# nvram get flag_try_sys1_failed
0
root@XiaoQiang:~# nvram get flag_try_sys2_failed
0`

NecronomiconCoding, 2023-11-16:
@remittor any idea?

remittor, 2023-11-17:
> Device now reboots and its still the same firmware as before.

I suspect that the bootloader doesn't like something. It's worth looking in the UART logs for the reason. And to get these logs you need to solder UART USB-TTL.

sophipl, 2024-02-21:
I tried flashing OpenWRT this way, and device rebooted with original firmware

Here is the log
[AX9000.log](https://github.com/openwrt-xiaomi/xmir-patcher/files/14362215/AX9000.log)

closed 2025-10-19 by remittor