openwrt-xiaomi/xmir-patcher #15: AX3000T - Error creating full backup

Vilsowski, 2024-02-04:
Router: Xiaomi AX3000T
Stock firmware version: 1.0.47

Hello, i'm getting a error when trying to create a full backup. I follow this steps.

```
Select: 2

device_name = RD03
rom_version = 1.0.47 release
mac address = 00:26:xx:xx:xx:xx
Current CountryCode = CN
Enter device WEB password: ********
Run SSH server on port 22 ...
#### SSH server are activated! ####
```

After that:

```
Select: 4

Download file: "/tmp/dmesg.log" ....
Download file: "/tmp/mtd_list.txt" ....
Download file: "/tmp/mtd_info.txt" ....
Download file: "/tmp/mtd_fdt.txt" ....
Download file: "/tmp/kcmdline.log" ....
Download file: "/tmp/dmesg.log" ....
Download file: "/tmp/mtd_list.txt" ....
Download file: "/tmp/mtd_info.txt" ....
Download file: "/tmp/mtd_fdt.txt" ....
Full backup creating...

ERROR: SSH execute command timed out! CMD: "dd if=/dev/mtd0 of=/tmp/mtd_dump.bin"
```

After that the router reboots.

Then i tried using choice 8 - 4

```
Choice: 4

Download file: "/tmp/dmesg.log" ....
Download file: "/tmp/mtd_list.txt" ....
Download file: "/tmp/mtd_info.txt" ....
Download file: "/tmp/mtd_fdt.txt" ....
Download file: "/tmp/kcmdline.log" ....
Download file: "/tmp/dmesg.log" ....
Download file: "/tmp/mtd_list.txt" ....
Download file: "/tmp/mtd_info.txt" ....
Download file: "/tmp/mtd_fdt.txt" ....
   0 > addr: 0x00000000  size: 0x08000000  name: "spi0.0"
   1 > addr: 0x00000000  size: 0x00100000  name: "BL2"
   2 > addr: 0x00100000  size: 0x00040000  name: "Nvram"
   3 > addr: 0x00140000  size: 0x00040000  name: "Bdata"
   4 > addr: 0x00180000  size: 0x00200000  name: "Factory"
   5 > addr: 0x00380000  size: 0x00200000  name: "FIP"
   6 > addr: 0x00580000  size: 0x00040000  name: "crash"
   7 > addr: 0x005C0000  size: 0x00040000  name: "crash_log"
   8 > addr: 0x00600000  size: 0x02200000  name: "ubi"
   9 > addr: 0x02800000  size: 0x02200000  name: "ubi1"
  10 > addr: 0x04A00000  size: 0x02000000  name: "overlay"
  11 > addr: 0x06A00000  size: 0x00C00000  name: "data"
  12 > addr: 0x07600000  size: 0x00040000  name: "KF"

Enter partition name or mtd number: 0

ERROR: SSH execute command timed out! CMD: "dd if=/dev/mtd0 of=/tmp/mtd_dump.bin"
```
And router reboots again.
I had the same issue with 8, 9, 10 and 11 partitions but without reboot. It's possible to be some kind of protection?

Gprsk, 2024-08-06:
This is just a timeout problem. Seems like the AX3000T is not powerful enough to finish running the command before Xmir Patcher gives up.

Quick workaround is to go into the code and manually set a (large) timeout value.

You alter file `gateway.py`, on line 719.
Add a new line after this one, changing the "timeout" variable to any amount of seconds:
`def run_cmd(self, cmd, msg = None, timeout = None, die_on_error = True):`
`   timeout = 30`

This is for 30 seconds, but you can put 45 or more just to be sure. Even a slow command shouldn't take more than 1 minute to run.

By doing this I could perform a full backup of all partitions, no problem.

ortegafernando, 2024-08-12:
Hi, I am with an usb c to rj45/lan adapter. (I think it is 100Mb not 1000Mb)

I am trying also to performa full backup of a RD03 AX3000T.

Not luck even with these changes:

Try change line 71 in create_backup.py from:
		  
```gw.run_cmd("dd if=/dev/mtd{id} of={o}".format(id=pid, o=fn_remote))```

to
		 
```gw.run_cmd("dd if=/dev/mtd{id} of={o}".format(id=pid, o=fn_remote), timeout=60)```

set timeout to 60 sec or more
and line 99 change to

```ret = gw.run_cmd(cmd, timeout=60, die_on_error = False)```

You alter file gateway.py, on line 719.
Add a new line after this one, changing the "timeout" variable to any amount of seconds:

```
def run_cmd(self, cmd, msg = None, timeout = None, die_on_error = True):
timeout = 30
```

What do you think is my problem?

```
device_name = RD03
rom_version = 1.0.47 release
mac address = 00:e0:4c:68:07:93
Current CountryCode = CN
Run SSH server on port 22 ...
#### SSH server are activated! ####

Select: 4

Download file: "/tmp/dmesg.log" ....
Download file: "/tmp/mtd_list.txt" ....
Download file: "/tmp/mtd_info.txt" ....
Download file: "/tmp/mtd_fdt.txt" ....
Download file: "/tmp/kcmdline.log" ....
Download file: "/tmp/dmesg.log" ....
Download file: "/tmp/mtd_list.txt" ....
Download file: "/tmp/mtd_info.txt" ....
Download file: "/tmp/mtd_fdt.txt" ....
Full backup creating...

ERROR: SSH execute command timed out! CMD: "dd if=/dev/mtd0 of=/tmp/mtd_dump.bin"
```

Gittson, 2024-10-29:
This is not due to timeout issue.  It is due to not enough space at /tmp during the image dumps.

EveElseIf, 2024-12-31:
I think there's not enough space at /tmp. I tried these 2 methods and they both worked.

1. using gzip when executing `dd` to compress the file, `dd if=/dev/mtd0 | gzip > /tmp/mtd0.bin.gz`
2. using netcat for directly copying mtd files to remote computer, the command is like `cat /dev/mtd0 | nc 192.168.31.57 12345`

The script may need to additionally provide one of these 2 methods to prevent the error.

2435043xia, 2025-01-27:
Based on the two answers above, you can modify line 71 in create_backup.py to: gw.run_cmd("dd if=/dev/mtd{id} | gzip > {o}".format(id=pid, o=fn_remote), timeout=60), and then rename the downloaded bin file to mtd0.bin.gz.

remittor, 2025-01-30:
Fixed https://github.com/openwrt-xiaomi/xmir-patcher/commit/03f5aabba7a0c029448e6f04634bee98db8ad06c

closed 2025-01-30 by remittor