openwrt-xiaomi/xmir-patcher #18: AX1800 和 AX3000解锁ssh失败，telnet和sftp都开启成功

LemonZuo, 2024-03-22:
device_name = RM1800
rom_version = 1.0.394 release
mac address = 00:0c:29:36:a3:39
CountryCode = CN
Enter device WEB password: ********
Enable smartcontroller scene executor ...
Wait smartcontroller activation ...
Unlock dropbear service ...
Unlock SSH server ...
Set password "root" for root user ...
Enabling dropbear service ...
Run SSH server on port 22 ...
Test SSH connection to port 22 ...

WARNING: SSH server not responding (IP: 192.168.31.246)

Unlock TelNet server ...
Run TelNet server on port 23 ...

#### TelNet server are activated! ####
#### FTP server are activated! ####

==========================================================

kissesu, 2024-11-24:
一样的问题🤣

qq11206246, 2025-01-03:
型号：小米万兆路由器(be10000)
device_name = RC01                                                                                                      
rom_version = 1.1.27 release                                                                                            
mac address = 00:e0:4c:68:20:b1                                                                                         
CountryCode = CN                                                                                                        
Enter device WEB password: a123456789                                                                                     
Detect using xiaoqiang "hackCheck" fix ;-)                                                                              
Enable smartcontroller scene executor ...                                                                               
Wait smartcontroller activation ...                                                                                     
                                                                                                                        
ERROR: Exploit "smartcontroller" not working!!!

SmartController 功能在大部分场景下默认处于禁用状态,需要激活它才行。

atolismesh, 2025-01-04:
> device_name = RC01
> rom_version = 1.1.27 release

Wow! Where do you get this new rom version (1.1.27) ? Latest version on Xiaomi site is 1.1.16 .  XMiR-Patcher works well with 1.1.16. 
Please give link to new rom version (1.1.27) for testing XMiR-Patcher

remittor, 2025-01-05:
> ERROR: Exploit "smartcontroller" not working!!!

Try downgrade to 1.1.16 firmware

> Please give a link to the new rom version (1.1.27) for testing XMiR-Patcher

http://cdn.cnbj1.fds.api.mi-img.com/xiaoqiang/rom/rc01/miwifi_rc01_firmware_9b62e_1.1.27.bin

qq11206246, 2025-01-05:
> > 错误：利用“smartcontroller”不起作用!!
> 
> 尝试降级到 1.1.16 固件
> 
> > 请提供新 rom 版本 （1.1.27） 的链接以测试 XMiR-Patcher
> 
> http://cdn.cnbj1.fds.api.mi-img.com/xiaoqiang/rom/rc01/miwifi_rc01_firmware_9b62e_1.1.27.bin

无法降级了。降级失败。(Unable to downgrade. Downgrade failed.)

remittor, 2025-01-05:
> Unable to downgrade. Downgrade failed.

Try downgrade via bootloader (tftp).