openwrt-xiaomi/xmir-patcher #19: BE3600

jdmpower, 2024-04-05:
Hello
Will there be an update for Xiaomi Miwifi BE3600 SSH access?
BR

pitiriguisvi, 2024-06-12:
+1 also interested

gelado, 2024-06-24:
Interested too :)

popos123, 2024-07-08:
im too BE3600 and BE5000

lem0ke, 2024-07-20:
im also

VVi3ard, 2024-07-24:
+1 also interested

GJCav, 2024-09-11:
This post may be helpful:

https://www.right.com.cn/forum/thread-8395205-1-1.html

atolismesh, 2024-09-11:
@GJCav 
right.com.cn is inaccessible  without Mainland Chaina QQ account. 
Can you post the script from right.com.cn here?

aqiyuyue, 2024-09-11:
curl -X POST http://192.168.31.1/cgi-bin/luci/;stok=xxx/api/xqsystem/start_binding -d "uid=1234&key=1234'%0Anvram%20set%20ssh_en%3D1'"curl -X POST http://192.168.31.1/cgi-bin/luci/;stok=xxx/api/xqsystem/start_binding -d "uid=1234&key=1234'%0Anvram%20commit'"curl -X POST http://192.168.31.1/cgi-bin/luci/;stok=xxx/api/xqsystem/start_binding -d "uid=1234&key=1234'%0Ased%20-i%20's%2Fchannel%3D.*%2Fchannel%3D%22debug%22%2Fg'%20%2Fetc%2Finit.d%2Fdropbear'"curl -X POST http://192.168.31.1/cgi-bin/luci/;stok=xxx/api/xqsystem/start_binding -d "uid=1234&key=1234'%0A%2Fetc%2Finit.d%2Fdropbear%20start'"

GJCav, 2024-09-14:
> @GJCav right.com.cn is inaccessible without Mainland China QQ account. Can you post the script from right.com.cn here?

Sorry for the delay. [aqiyuyue](https://github.com/openwrt-xiaomi/xmir-patcher/issues/19#issuecomment-2343444062) gave the core commands. A more readable version in `powershell` is shown here:

``` powershell
$stok='<magic token shown in the URL after loggin in to the web admin panel>'
$ip='<ip of your router>'

curl -X POST "http://$ip/cgi-bin/luci/;stok=$stok/api/xqsystem/start_binding" -d "uid=1234&key=1234'%0Anvram%20set%20ssh_en%3D1'"
curl -X POST "http://$ip/cgi-bin/luci/;stok=$stok/api/xqsystem/start_binding" -d "uid=1234&key=1234'%0Anvram%20commit'"
curl -X POST "http://$ip/cgi-bin/luci/;stok=$stok/api/xqsystem/start_binding" -d "uid=1234&key=1234'%0Ased%20-i%20's%2Fchannel%3D.*%2Fchannel%3D%22debug%22%2Fg'%20%2Fetc%2Finit.d%2Fdropbear'"
curl -X POST "http://$ip/cgi-bin/luci/;stok=$stok/api/xqsystem/start_binding" -d "uid=1234&key=1234'%0A%2Fetc%2Finit.d%2Fdropbear%20start'"
```

you may need to modify the script for linux environment.

After starting the ssh server (dropbear in the case of XiaoQiang), the password of root can be calculate by https://miwifi.dev/ssh

jdmpower, 2024-09-14:
> Sorry for the delay. [aqiyuyue](https://github.com/openwrt-xiaomi/xmir-patcher/issues/19#issuecomment-2343444062) gave the core commands. A more readable version in `powershell` is shown here: [...]

I was very happy to open SSH access but.

although I edited the network and port_map file, the 8021q vlanid definition (such as be7000,be6500pro) does not work on the pppoe side, I hope someone can help, I wonder if the switch is not located in the SOC.

GJCav, 2024-09-16:
@jdmpower I have limited knowledge about the Ethernet and its physical layer specifications. My goal in enabling ssh was to set custom IPv6 firewall rules and port forwarding, which operate at the network and transport layer. So I cannot help you. But I think these resources may be helpful:

[This post](https://www.acwifi.net/26827.html) provides a detailed teardown of Xiaomi BE3600. Though it is written in Chinese, you can use a web translation tool to read it. Here is a brief summary:

[image]

As you can see, the switch is not intergrated into the SoC.

The switch chip, YT8215S, is manufactured by a Chinese company called 裕太微电子, and you can find the chip is listed [here](https://www.motor-comm.com/product/switching-chip). Although the official website does not provide the datasheet, I found [this](https://item.szlcsc.com/6834649.html) from [嘉立创](https://www.jlc.com/), a well-known Chinese PCB manufacturer.

According to the datasheet of YT8215S, it supports lite-L2-management (the translation might not be perfect), so I believe the 8021q vlanid definition is supported by the switch, but may be not supported by the official XiaoQiang system.

If you have time and enthusiasm to dig deeper into the system and the underlying chip-2-chip PIN connections, you may achieve your goal. So, good luck!

pitiriguisvi, 2024-09-16:
does anyone managed to get VLAN working in BE3600? I´ve edited network file:

config interface 'internet_wan'
	option ifname 'eth1.6'

config interface 'wan'
	option proto 'pppoe'
	list dns '192.168.2.7'
	list dns '1.1.1.1'
	option peerdns '0'
	option username 'adsl'
	option mru '1480'
	option special '0'
	option password 'adsl'
	option ifname 'eth1.6'

But nothing works (vlan=6)

leitao1019, 2024-09-20:
Miwifi BE3600 2.5G
system version: 1.0.68 
run above http POST command return error code as below:
{"code":1523}

Ericnunes77, 2024-09-20:
My device just stay with orange light. Don´t appear Ip number...Please guys, help me....how i fix that? umbrick?

alessandrotrevi, 2024-09-27:
> Sorry for the delay. [aqiyuyue](https://github.com/openwrt-xiaomi/xmir-patcher/issues/19#issuecomment-2343444062) gave the core commands. A more readable version in `powershell` is shown here: [...]

I managed to activate SSH with this script (thank you) but I can't access via SSH.
It keep saying my password is wrong.

I inserted the SN in the link you posted and I got a password, it seems to be wrong though. Is there any other way to find the password? Or maybe I'm doing something wrong?

CristianZatt, 2024-09-29:
> Sorry for the delay. [aqiyuyue](https://github.com/openwrt-xiaomi/xmir-patcher/issues/19#issuecomment-2343444062) gave the core commands. A more readable version in `powershell` is shown here: [...]

SSH worked, but the files that lock the channels are either uneditable or get restored to default every restart. Also tryed to change region to unlock but it seems the only region available is CN. Has anyone been able to unlock 100 - 144 channels?

alessandrotrevi, 2024-09-29:
Ssh worked for me too, but I cannot change vlan tag or region. I edited network file but it doesn’t seem to work.

any ETA for Openwrt for this router? I’m sure a custom firmware would fix all these problems

Ericnunes77, 2024-10-09:
Guys, someone can help me? my device just stay in orange light. I reset it but nothing fix that. How i do a super hard reset? reinstal a firmware can fix that?

MaCK1e, 2024-11-05:
> does anyone managed to get VLAN working in BE3600? I´ve edited network file:
>
> config interface 'internet_wan' option ifname 'eth1.6'
>
> config interface 'wan' option proto 'pppoe' list dns [...]

Hello! you can need edit only the port_map File

this config works for me with the VLAN 881:

config port '4'
	option speed '2.5G'
	option link_mode '0'
	option base_iface 'eth1'
	option type 'eth'
	option label '4(2.5G)'
	option ifname 'eth1.881'
	option service 'wan'

pitiriguisvi, 2024-11-05:
@MaCK1e how would port_map file be if i want to use LAN1 as wan port?

caicob, 2024-12-18:
https://www.kjfx.cc/647.html

pitiriguisvi, 2025-01-10:
is there a way to create an overlay in be3600 (such miwifi_overlay in ax3600) to intall other packages?

vagoals, 2025-02-04:
I ssh done I done't flie OPERWRT BE3600 pls

Raphitpt, 2025-02-12:
No update for openwrt firmware ?

scriptsboy, 2025-03-13:
Is there any OPENWRT firmware available for BE3600?

closed 2025-10-19 by remittor