openwrt-xiaomi/xmir-patcher #20: "Xiaomi Mi Wi-Fi device not found" despite model seems to be supported

wschlich, 2024-04-11:
According to `xqmodel.py`, the model `CB0401` should be supported.
However, when I try to run menu item 2, I get the following error:
```
ERROR: Xiaomi Mi Wi-Fi device not found (IP: 192.168.31.1)
```
The device currently runs MiWifi firmware 3.0.100 (2023/10/10).

kejlyCZ, 2025-01-16:
Same problem with Xiaomi AX3000T international running MiWiFi 1.0.90 (should be regognized as RD23), but router not found.

remittor, 2025-01-16:
> Same problem with Xiaomi AX3000T international running MiWiFi 1.0.90 (should be regognized as RD23), but router not found.

downgrade firmware via TFTP or MiWiFiTool

kejlyCZ, 2025-01-16:
Thanks for advices. Unfortunately MiWiFiRepairtool doesn't work at all (either detecting wrong IP addresses, or doing nothing), I've tried tftp and also no success (tftp on Windows is pain in the *ss).  Thanks again, I'll wait, until ssh/OpenWrt is available for new firmware, or run native firmware.

wschlich, 2025-06-11:
Hey @remittor!

xmir-patcher still seems to be unable to connect to my CB0401:
```
ERROR: Xiaomi Mi Wi-Fi device not found (IP: 192.168.31.1)
```

The router status page shows:
```
Model: CB0401(HW0.1)
System version: MiWiFi 3.0.116(2024/04/03)
```

The device *is* reachable under the IP address 192.168.31.1.

What's wrong? :-)

Cheers,
Wolfram

remittor, 2025-06-16:
You probably haven't completed the initial setup of your router.

wschlich, 2025-06-23:
@remittor What exactly do you mean by "initial setup"?

remittor, 2025-10-19:
Fixed after adding support https