openwrt-xiaomi/xmir-patcher #22: AX300t installed firmware now ip address not being signed

hassanm12, 2024-04-25:
Select: 2

device_name = RD03
rom_version = 1.0.47 release
mac address = 00:e0:4c:68:0b:f2
Current CountryCode = CN
Run SSH server on port 22 ...
#### SSH server are activated! ####

==========================================================

Xiaomi MiR Patcher

 1 - Set IP-address (current value: 192.168.1.1)
 2 - Connect to device (install exploit)
 3 - Read full device info
 4 - Create full backup
 5 - Install EN/RU languages
 6 - Install Breed bootloader
 7 - Install firmware (from directory "firmware")
 8 - {{{ Other functions }}}
 9 - [[ Reboot device ]]
 0 - Exit

Select: 7

device: "RD03"
img_write = True
Image files in directory "firmware/":
  "firmware/openwrt-mediatek-filogic-xiaomi_mi-router-ax3000t-initramfs-factory.ubi"
Download file: "/tmp/dmesg.log" ....
Download file: "/tmp/mtd_list.txt" ....
Download file: "/tmp/mtd_info.txt" ....
Download file: "/tmp/mtd_fdt.txt" ....
Download file: "/tmp/kcmdline.log" ....
Parse all images...
UBI: filetype: b'UBI#'
UBI: Decoding UBIFS...
UBI:   volume: "kernel"          size: 7491584
parse_ubifs = 1
FIT size = 0x715034 (7252 KiB)
FIT: name = "ARM64 OpenWrt FIT (Flattened Image Tree)"
FIT: def_cfg: "config-1"
FIT: def_cfg desc = "OpenWrt xiaomi_mi-router-ax3000t"
FIT: model = "xiaomi,mi-router-ax3000t"
FIT: def_fdt: "fdt-1"
FDT: desc = "ARM64 OpenWrt xiaomi_mi-router-ax3000t device tree blob"
FDT: type = "flat_dt"
FDT: arch = "arm64"
KRN: desc = "ARM64 OpenWrt Linux-6.1.86"
KRN: type = "kernel"
KRN: arch = "arm64"
KRN: compression = "lzma"
KRN: data = 4147826 bytes
FDT: compatible = ['xiaomi,mi-router-ax3000t', 'mediatek,mt7981']
FDT: model = "Xiaomi Mi Router AX3000T"
FDT: dt_part: ['/soc/spi@1100a000/flash@0/partitions']
FIT: Founded "initrd-1" node
FIT: initrd image name: "ARM64 OpenWrt xiaomi_mi-router-ax3000t initrd"
fw_img: 8448 KiB | kernel: 7252 KiB | rootfs: 3176 KiB
Download file: "/tmp/bl_BL2.bin" ....
Download file: "/tmp/bl_FIP.bin" ....
Download file: "/tmp/env_Nvram.bin" ....
Download file: "/tmp/env_Bdata.bin" ....
Download file: "/tmp/env_BL2.bin" ....
current flag_boot_rootfs = 0
install_method = 400
--------- prepare command lines -----------
fw_img: 8448 KiB | kernel: 7252 KiB | rootfs: 3176 KiB
------------- flash images -------------
Upload file: "tmp/fw/fw_img.bin" ....
Run scripts for change NVRAM params...
Boot from firmware [1] activated.
Writing firmware image to addr 0x02800000 ...
  mtd -e "ubi1" write "/tmp/fw_img.bin" "ubi1"
The firmware has been successfully flashed!
Send command "reboot" via SSH/Telnet ...
Forced REBOOT activated!

I installed 2 and then 7, it rebooted, now i cant connect to it. if i do IPconfig, the ipaddress it shows is
   Autoconfiguration IPv4 Address. . : 169.254.210.213
   Subnet Mask . . . . . . . . . . . : 255.255.0.0
   Default Gateway . . . . . . . . . :
 
Did I just brick my device?

utajum, 2024-04-25:
Did you try different ports?

pedroknup, 2025-01-04:
Same here. Any luck?

atolismesh, 2025-01-04:
Assign temporary fixed ip address 192.168.1.10/255.255.255.0 to computer.

closed 2025-10-19 by remittor