openwrt-xiaomi/xmir-patcher #27: Ax 3000 ra80v2

JuanManuelRomeroGarcia, 2024-06-09:
Ax 3000 ra80v2 no support

AndiTails, 2024-06-12:
Seconded. This is the output:

```
device_name = RA80V2
rom_version = 1.0.15 release
mac address = <redacted>
CountryCode = CN
Enter device WEB password: <redacted>
Detect using xiaoqiang "hackCheck" fix ;-)
Enable smartcontroller scene executor ...
Wait smartcontroller activation ...

ERROR: Exploit "smartcontroller" not working!!!
```

CN-MRZZJ, 2024-06-13:
third
device_name = RA80V2
rom_version = 1.0.15 release
mac address = 74:97:79:e4:17:25
CountryCode = CN
Detect using xiaoqiang "hackCheck" fix ;-)
Enable smartcontroller scene executor ...
Wait smartcontroller activation ...

ERROR: Exploit "smartcontroller" not working!!!

1302557841, 2024-06-26:
device_name = RA80V2
rom_version = 1.0.15 release
mac address = 18:c0:4d:f8:07:cd
CountryCode = CN
Detect using xiaoqiang "hackCheck" fix ;-)
Enable smartcontroller scene executor ...
Wait smartcontroller activation ...

ERROR: Exploit "smartcontroller" not working!!!

derectim, 2024-09-24:
device_name = RA80V2
rom_version = 1.0.15 release
mac address = <redacted>
CountryCode = CN
Enter device WEB password: <redacted>
Detect using xiaoqiang "hackCheck" fix ;-)
Enable smartcontroller scene executor ...
Wait smartcontroller activation ...

ERROR: Exploit "smartcontroller" not working!!!

remittor, 2024-09-24:
Add support for RA80V2

JuanManuelRomeroGarcia, 2024-09-27:
After enabling ssh it loses wifi connection

hiagopdutra, 2025-03-05:
@bed415 did you make something to work with you? I tried here and I got the same error as other folks