openwrt-xiaomi/xmir-patcher #31: SSH error

MaginaHou, 2024-08-08:
[image]
I am using a Xiaomi 10 Gigabit router, and in the early stages, SSH can be enabled through this software. After opening it, I tried flashing the device a few more times. Now, when I click on 2, it will report an error as shown in the picture. Could you please help me identify where the problem lies.

Polarisazz, 2024-09-19:
he

> Same here 同样在这里

Hey bro,I've found a solution.Just use telnet to access the router,enter "sed -i '/flg_ssh=`nvram get ssh_en`/{:loop; N; /\n.*channel=`\/sbin\/uci get \/usr\/share\/xiaoqiang\/xiaoqiang_version.version.CHANNEL`\n.*return 0\n.*fi/!b loop; d}' /etc/init.d/dropbear",enter "/etc/init.d/dropbear restar".Then you can use SSH.

Polarisazz, 2024-09-20:
Hey,I've found a solution.Just use telnet to access the router,enter "sed -i '/flg_ssh=nvram get ssh_en/{:loop; N; /\n.*channel=\/sbin\/uci get \/usr\/share\/xiaoqiang\/xiaoqiang_version.version.CHANNEL\n.*return 0\n.*fi/!b loop; d}' /etc/init.d/dropbear",enter "/etc/init.d/dropbear restar".Then you can use SSH.

lan493, 2024-11-10:
> Hey,I've found a solution.Just use telnet to access the router,enter "sed -i '/flg_ssh=nvram get ssh_en/{:loop; N; /\n.*channel=/sbin/uci get [...]
不对啊老哥，telnet命令之后没有作用，还是你哪里打错了