openwrt-xiaomi/xmir-patcher #46: xiaomi ac2100

morkvk, 2024-10-22:
error
plz help 

device: "R2100"
img_write = True
Image files in directory "firmware/":
  "firmware/openwrt-ramips-mt7621-xiaomi_mi-router-ac2100-initramfs-kernel.bin"
Download file: "/tmp/dmesg.log" ....
Download file: "/tmp/mtd_list.txt" ....
Download file: "/tmp/mtd_info.txt" ....
Download file: "/tmp/mtd_fdt.txt" ....

ERROR: Partition list is empty! (solution: disable all WiFi modules and reboot device)

wersonit, 2025-09-16:
I had the same problem on many other routers too. The only way to install Openwrt on Xiaomi Router AC2100, Xiaomi 3G, Xiaomi 4 etc. is to use BREED for R3G. There are many BREED versions, you need to find the one that supports flashing kernel1 and rootfs0.

MicroDeveci, 2025-10-16:
> I had the same problem on many other Xiaomi routers too. The only way to install Openwrt on Xiaomi Router AC2100, Xiaomi 3G, Xiaomi 4 etc. is to use BREED for R3G. There are many BREED versions, you need to find the one that supports flashing kernel1 and rootfs0. I used BREEDEnter which came with wrong BREED version (doesn't support kernel1/rootfs0) and used this wrong BREED to upload correct BREED using Bootloader section upgrade.

I think this error was caused by limited ram ,my rm2100(redmi ac2100) only 128mb ram ,when i manually copy partition to /temp and error still