openwrt-xiaomi/xmir-patcher #57: leaked shared memory object warning message

fdauti, 2024-12-24:
Hi, I'm using Centos10 on a vmware vm, connected through wifi. 
I get a warning about leaked-shared memory object as in the image below. I only tested on section 1 and 0.
Is there a concern about the patch failing with the other sections?
[image]

alekam, 2025-01-04:
Mac OS Ventura 13.7 with python 3.9 from Homebrew the same

aosucas499, 2025-01-08:
Me too. Ubuntu 20.04. I only tried to reboot the router. Scared to try the exploit. (BE6500)

fdauti, 2025-01-08:
> Me too. Ubuntu 20.04. I only tried to reboot the router. Scared to try the exploit. (BE6500)

On Windows with the AX3200 it went smooth on my side, so there shouldn't be problems with Windows

remittor, 2025-01-28:
Fixed https://github.com/openwrt-xiaomi/xmir-patcher/commit/812649c8baee6270f0fe9e40b265d3f955d6d795

closed 2025-01-28 by remittor