openwrt-xiaomi/xmir-patcher #58: Firmware for cb401v2 anyone?

Moonklif, 2025-01-02:
Someone have a Firmware file for Xiaomi cb401v2

remittor, 2025-03-06:
Not founded fw :-(

remittor, 2025-10-15:
I managed to find the following firmware for CB0401V2:
* http://cdn.awsde0-fusion.fds.api.mi-img.com/xiaoqiang/rom/cb0401v2/miwifi_cb0401v2_R03A02-FOTA_firmware_d616a_3.0.29_INT_ispver-3.0.2.bin
* http://cdn.awsde0-fusion.fds.api.mi-img.com/xiaoqiang/rom/cb0401v2/miwifi_cb0401v2_R03A04-FOTA_firmware_3f953_3.0.34_INT_ispver-3.0.3.bin
* http://cdn.awsde0-fusion.fds.api.mi-img.com/xiaoqiang/rom/cb0401v2/miwifi_cb0401v2_R03A04-FOTA_firmware_d27f4_3.0.54_INT_ispver-3.0.5.bin
* http://cdn.awsde0-fusion.fds.api.mi-img.com/xiaoqiang/rom/cb0401v2/miwifi_cb0401v2_R03A04-FOTA_firmware_826fe_3.0.57_INT_ispver-3.0.6.bin

It is not yet possible to find a valid name for `3.5.x` firmwares.

NikoTeslaWU, 2026-01-05:
http://cdn.cnbj1.fds.api.mi-img.com/xiaoqiang/rom/cb0401v2/miwifi_cb0401v2_R03A05-FOTA_all_287a2_3.0.73_INT_ispver-3.0.7.bin

NikoTeslaWU, 2026-01-05:
{"ispver":"3.0.7","downloadUrl":"http://cdn.cnbj1.fds.api.mi-img.com/xiaoqiang/rom/cb0401v2/miwifi_cb0401v2_R03A05-FOTA_all_287a2_3.0.73_INT_ispver-3.0.7.bin","changelogUrl":"https://cdn.azams0.fds.api.mi-img.com/miwifi/d2fe24a8-0331-4d72-93c5-1e5cd0c71f9f.html","code":0,"buildTime":"2025/09/18","changeLog":"","version":"3.0.73","weight":1,"status":{"status":0,"percent":0},"fileSize":151767136,"needUpdate":1,"fullHash":"4275a8aaf2ecad559574d3c3c09287a2"}

crerser, 2026-01-05:
> > > I have a problem. My router version is 3.0.1, but I can't upgrade to 3.0.7 from the website; it keeps giving me an error.
> > > <img alt="Image" width="873" height="342" src="https://private-user-images.githubusercontent.com/58394590/531985969-758a17c0-9590-4977-84e4-00877c9fe87e.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Njc2MzA0NDMsIm5iZiI6MTc2NzYzMDE0MywicGF0aCI6Ii81ODM5NDU5MC81MzE5ODU5NjktNzU4YTE3YzAtOTU5MC00OTc3LTg0ZTQtMDA4NzdjOWZlODdlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNjAxMDUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjYwMTA1VDE2MjIyM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWFjOWUzYjc5NDViZjhiNDdlNjRkZjgxZDRhYjlkOGQ4YWI4MDE4OGUyZDE1MGE4ZmM2ODRkNWJkYmY4MTE3YzUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.DQN-wlSgRe2tHfQZXhYESsyitPwPnbgWjTB6xbDc5Nk"> <img alt="Image" width="793" height="343" src="https://private-user-images.githubusercontent.com/58394590/531985995-72da4f0a-e8c9-4a13-9733-d4df465c6cf7.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Njc2MzA0NDMsIm5iZiI6MTc2NzYzMDE0MywicGF0aCI6Ii81ODM5NDU5MC81MzE5ODU5OTUtNzJkYTRmMGEtZThjOS00YTEzLTk3MzMtZDRkZjQ2NWM2Y2Y3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNjAxMDUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjYwMTA1VDE2MjIyM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWM1MWJmZDUwNmU5NGQ0MzRkZWI2YzMxNjRlOTE5NmQ3ZWViYWMzMmY2NjU3MmRmYTZjZDYxM2MxYjg4ZmJkNjAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.0jQdcs9SAxEAEkhOrMV7rs9U_nW-L2Zc4unuMicJ0Mg">
> > > How to force an upgrade using TTL or SSH?
> > 
> > 
> > did you tried xmir-patcher (update from firmware folder?
> 
> i try but got error.
> <img alt="Image" width="893" height="591" src="https://private-user-images.githubusercontent.com/58394590/532038387-e85fd15b-3226-446e-8ea8-13b8c19a8490.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Njc2MzMzNzksIm5iZiI6MTc2NzYzMzA3OSwicGF0aCI6Ii81ODM5NDU5MC81MzIwMzgzODctZTg1ZmQxNWItMzIyNi00NDZlLThlYTgtMTNiOGMxOWE4NDkwLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNjAxMDUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjYwMTA1VDE3MTExOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTA4YWQ4OWFjNTkxMTdmNmZkZmIyZTFhNjE5NzIxM2VkMjkzMmQzZjQ0YWFiZTJiMTU0YTUzOTYxYTE4ODRhOTAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.ImBFOjSbJQmMvJftKClQSYj0NjHUKGN1sk5WKdT4QK0"> <img alt="Image" width="669" height="291" src="https://private-user-images.githubusercontent.com/58394590/532038463-f892f41d-bfba-40bf-b2bf-389e76b8bbb3.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Njc2MzMzNzksIm5iZiI6MTc2NzYzMzA3OSwicGF0aCI6Ii81ODM5NDU5MC81MzIwMzg0NjMtZjg5MmY0MWQtYmZiYS00MGJmLWIyYmYtMzg5ZTc2YjhiYmIzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNjAxMDUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjYwMTA1VDE3MTExOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTJiMzIyMDkwYjFlNzg0ZTQ4OTBmZTljMTc2OTk1ZDY0NWE2MzM2MWZmYWI5NWMwOGNjOWFiZWVkY2U3ZDRjNWImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.8qIZu7Rd8wMMJv3cQ0Q27PNGLjNM-ZFXXuHY4hPImJk">

try nº 2 - 3 - 4 - 7

NikoTeslaWU, 2026-01-06:
> try nº 2 - 3 - 4 - 7

same again

joaoramosipl, 2026-01-06:
> same again

I'm having exactly the same issue...

crerser, 2026-01-07:
Is there an online forum in Chinese that discusses this router? I'm trying to understand these (v2) routers better, but unfortunately I always end up in the same places. They are known as ax5400, magenta or xiaomi cpe pro 5g, cb0401.

svenvogel, 2026-02-08:
now Version: MiWiFi 3.0.8(2026/01/08) over ota

KittyBua, 2026-04-21:
@svenvogel Okay, thanks! I also have a T-Mobile phone with version 3.0.8, all original and branded!

That means I can update to 3.0.85 directly through the web interface using local installation! Thanks!

Kreal-exe, 2026-09-14:
If xmir-patcher/Python isn't required for what you need — I built a self-hosted toolkit that opens SSH/root access natively (same default-password mechanism, no Python dependency) and adds a local web GUI for Wi-Fi/cellular band tuning, firmware downgrade, and a permanent 5G SA + extra-band unlock. Covers both CB0401 and CB0401V2: https://github.com/Kreal-exe/cb0401-tune-control

KittyBua, 2026-09-22:
> Wenn xmir-patcher/Python für das, was Sie benötigen, nicht erforderlich ist — Ich habe ein selbst gehostetes Toolkit erstellt, das den SSH-/Root-Zugriff nativ öffnet (gleicher Standardpasswortmechanismus, keine Python-Abhängigkeit), die Telemetrie-/Junk-Cron-Jobs entfernt, mit denen die Standard-Firmware standardmäßig nach Hause telefoniert, und eine lokale Web-GUI für Wi-Fi/Mobilfunkband-Tuning, Firmware-Downgrade, hinzufügt und eine permanente 5G SA + Extraband-Entsperrung. Deckt sowohl CB0401 als auch CB0401V2 ab: https://github.com/Kreal-exe/cb0401-tune-control

Wau thanks!

Aber es ist trotzdem irgendwie sehr verwirrend! 

Man muss ja viel machen oder 

Danke sehe es mir an 

!!!!!

Wow, thanks!

But it's still really confusing somehow!

You have to do a lot, right?

Thanks, I'll take a look at it.

!!!!!

KittyBua, 2026-09-22:
OK with ssh Look morning this gui 

Thanks 

I only need the band settings, though.

P.S. By the way, I ran it using Termux and Ubuntu with Go for the GUI.

You can also see the SSH key in the GUI.

Thanks, great job! 👍

[image]
[image]

gthb96, 2026-10-15:
@KittyBua @Kreal-exe 
I know you are quite active in another cb0401 project.
Anyone of you knows if it's possible to remove Magenta branding by flashing unbranded firmware?

KittyBua, 2026-10-16:
> 31\. [@KittyBua](https://github.com/KittyBua) [@Kreal-exe](https://github.com/Kreal-exe)
>      I know you are quite active in another cb0401 project.
>      Anyone of you knows if it's possible to remove Magenta branding by flashing unbranded firmware?

Hey 👋 unfortunately not—I’m not aware of anything!

That said, an acquaintance of mine has the same V2 model, but he doesn’t use a T-Mobile SIM card, and his menu is blue—so, no T-Mobile branding!

If he inserts a Magenta (T-Mobile) SIM card, it turns pink and the branding appears!

Maybe that helps you out a bit!

Other Sim in a Magenta V2

[image]

gthb96, 2026-10-16:
Thanks for the input!
Here's what I've noticed with my V2, and also with a friend's CB0401 (not the V2): without a SIM card, both devices show as unbranded, but as soon as any SIM is inserted, even one from A1 or Drei, they switch to Magenta branding.
Weird that it is different than what you observed.
Not that it is a problem tho.
I still wonder where the branding is stored tho (firmware, configuration...)