openwrt-xiaomi/xmir-patcher #61: telnetlib Module Removed in Python 3.13 - Compatibility Issue with Provided Code

fretory, 2025-01-18:
**Environment Information:**
- Python Version: 3.13.1  
- Operating System: macOS  
- Command Run: `sh run.sh`  

**Observed Behavior:**  
When running the provided code, it attempts to import the `telnetlib` module, which results in the following error:  
```
Traceback (most recent call last):
  File "/Users/wangtao/Developer/xmir-patcher/menu.py", line 9, in <module>
    import gateway
  File "/Users/wangtao/Developer/xmir-patcher/gateway.py", line 23, in <module>
    import telnetlib
ModuleNotFoundError: No module named 'telnetlib'
```

**Reason for the Issue:**  
The `telnetlib` module was deprecated in Python 3.11 and has been removed entirely in Python 3.13, as per [[PEP 594](https://www.python.org/dev/peps/pep-0594/)](https://www.python.org/dev/peps/pep-0594/). Reference: [[Python 3.13 telnetlib documentation](https://docs.python.org/3.13/library/telnetlib.html)](https://docs.python.org/3.13/library/telnetlib.html).

**Steps Taken to Address the Issue:**
1. Verified the Python version (`Python 3.13.1`).
2. Reviewed the official documentation, which confirms the removal of `telnetlib` in Python 3.13.
3. Attempted to use an alternative library, [[telnetlib3](https://github.com/jquast/telnetlib3)](https://github.com/jquast/telnetlib3), as suggested by the Python documentation.

**Suggested Fix:**  
The current implementation should be updated to replace the use of `telnetlib` with a third-party library such as:
1. [[telnetlib3](https://github.com/jquast/telnetlib3)](https://github.com/jquast/telnetlib3) - Asynchronous Telnet client.

Would you consider updating the code to accommodate the removal of `telnetlib` in Python 3.13? Alternatively, could you provide guidance or recommendations for adapting the provided script for compatibility with Python 3.13?

Thank you for your work and support!

NotSBad, 2025-01-18:
https://github.com/openwrt-xiaomi/xmir-patcher/issues/60#issuecomment-2599538718

remittor, 2025-01-28:
Fixed https://github.com/openwrt-xiaomi/xmir-patcher/commit/0fd167ec76be27afa56cc8185c5900c154909f3e

closed 2025-01-28 by remittor