openwrt-xiaomi/xmir-patcher #64: Firmware for cb0401 anyone?

zlibtest, 2025-01-20:
When I tried to upgrade the system and received a prompt that it was 100% complete, the machine automatically entered the upgrade and restarted. However, after restarting, the system remained the old version (MiWiFi 1.0.4) and fell into the dilemma of not being able to upgrade. In addition, I also tried downloading firmware (https://cdn.awsde0-fusion.fds.api.mi-img.com/xiaoqiang/rom/cb0401/miwifi_cb0401_R01A11-FOTA_firmware_b173d_3.0.100_INT.bin) for uploading and upgrading or used xmir-patcher, but like online upgrading, it still stayed in the old version after restarting. So, could anyone help me?

remittor, 2025-03-06:
CB0401
https://cdn.awsde0-fusion.fds.api.mi-img.com/xiaoqiang/rom/cb0401/miwifi_cb0401_R01A13-FOTA_firmware_dce0f_3.0.116_INT.bin

zlibtest, 2025-03-06:
> CB0401 https://cdn.awsde0-fusion.fds.api.mi-img.com/xiaoqiang/rom/cb0401/miwifi_cb0401_R01A13-FOTA_firmware_dce0f_3.0.116_INT.bin

Hello, my dear friend, I have tried this, however once upgraded and the router automatically restarts, the system still displays the old version. No matter how I try to upgrade the new firmware, the system is still an old version, and I don't know what the problem is. Do you have any good suggestions?

zlibtest, 2025-03-06:
> CB0401 https://cdn.awsde0-fusion.fds.api.mi-img.com/xiaoqiang/rom/cb0401/miwifi_cb0401_R01A13-FOTA_firmware_dce0f_3.0.116_INT.bin

I also use xmir-patcher to upgrade, but it reminds me “guess_filetype Fatal: Could not determine file type."

[image]

remittor, 2025-03-06:
> I also use xmir-patcher to upgrade,

XMiR not support these firmwares. Sorry

zlibtest, 2025-03-06:
> > I also use xmir-patcher to upgrade,
> 
> XMiR not support these firmwares. Sorry

I see, thank you so much! Also, do you have any other suggestions for the difficulties I have encountered?

jexnator, 2025-04-07:
Are there plans to implement support for these firmwares?

Kreal-exe, 2026-09-14:
If xmir-patcher/Python isn't required for what you need — I built a self-hosted toolkit that opens SSH/root access natively (same default-password mechanism, no Python dependency) and adds a local web GUI for Wi-Fi/cellular band tuning, firmware downgrade, and a permanent 5G SA + extra-band unlock. Covers both CB0401 and CB0401V2: https://github.com/Kreal-exe/cb0401-tune-control