openwrt-xiaomi/xmir-patcher #74: ERROR: Exploits arn_switch/start_binding not working!!!

small1445, 2025-02-23:
device： redmi cr8808
rom_version：6.2.18
system：windows 11
question: The arn_switch/start_binding injection point is now defunct.

[image]

remittor, 2025-02-24:
> device： redmi cr8808
> rom_version：6.2.18

Downgrade firmware to 6.2.11 via MIWIFIRepairTool
https://cdn.cnbj1.fds.api.mi-img.com/xiaoqiang/rom/cr8808/miwifi_cr8808_firmware_9d216_6.2.11.bin

remittor, 2025-02-24:
Please try latest version

JuanManuelRomeroGarcia, 2025-02-24:
> Please try latest version

try:
gw = inited_gw
except NameError:
gw = create_gateway(die_if_sshOk = False)

inited_gw not defined error in connect5 and connect6

tuilakhanh, 2025-02-24:
> Please try latest version

Work fine now. Thank you.

agritux, 2025-02-27:
> Please try latest version

Xiaomi AX3000T same issue with latest version

[image]

agritux, 2025-02-27:
> Please try latest version

AX3000T same issue with latest version

[image]

remittor, 2025-02-28:
> AX3000T same issue with latest version

needed downgrade firmware to 1.0.49 via MiWifi Repair Tool  (or tftpd64)

agritux, 2025-02-28:
> > AX3000T same issue with latest version
> 
> needed downgrade firmware to 1.0.49 via MiWifi Repair Tool (or tftpd64) Example: https://forum.openwrt.org/t/openwrt-support-for-xiaomi-ax3000t/180490/3313

Thank you!

CalgarMA, 2025-03-05:
Hi! I am using the latest version of X MiR patcher for Xiaomi AX3000T (RD03, 1.0.84). I can't connect to the device, point 2. Do I need to roll back the firmware (for version 1.0.84 I don't think so)?
[image]

remittor, 2025-03-05:
> I am using the latest version of X MiR patcher for Xiaomi AX3000T (RD03, 1.0.84). I can't connect to the device, point 2.

Try latest version again.

closed 2025-03-06 by remittor