openwrt-xiaomi/xmir-patcher #76: Can't connect to Wi-Fi after patch

freefelt, 2025-03-06:
BE7000
Firmware: 1.1.16

```
mtd erase crash
reboot
```
- does not work.

remittor, 2025-03-06:
Я ваши мысли читать не умею. Нужно больше входной информации для ответа.

freefelt, 2025-03-06:
Хорошо. Сбрасываю роутер. Прохожу начальную настройку. К вайфаю все подключается без проблем. Запускаю патч, выбираю пункт 2 для установки эксплойта. Все устанавливается без проблем. Но уже в этот момент телефон отваливается от вайфая и потом к нему подключиться уже не получается. Сеть видно, пароль тот же (менять тоже пробовал), но не подключается. [Тут](https://github.com/openwrt-xiaomi/xmir-patcher/issues/13) предлагали команды ввести, они не помогают.

remittor, 2025-03-06:
Попробуйте в файле "connect.py" удалить эти строчки:
```
    hackCheck = gw.detect_hackCheck(update = True)
    if hackCheck:
        print(f'hackCheck version =', hackCheck)
```
И проверьте снова.
И желательно лог из консоли сюда копировать!

freefelt, 2025-03-06:
Не помогло. Роутер сбросил к заводским перед пробой.

```
==========================================================

Xiaomi MiR Patcher

 1 - Set IP-address (current value: 192.168.31.1)
 2 - Connect to device (install exploit)
 3 - Read full device info
 4 - Create full backup
 5 - Install EN/RU languages
 6 - Install Breed bootloader
 7 - Install firmware (from directory "firmware")
 8 - {{{ Other functions }}}
 9 - [[ Reboot device ]]
 0 - Exit

Select: 2

device_name = RC06
rom_version = 1.1.16 release
mac_address = a8:a1:59:95:84:a7
CountryCode = CN
Enter device WEB password: teanasLife33
WARN: Exploits "arn_switch/start_binding" not working!!!
Detect using xiaoqiang "hackCheck" fix ;-)
Enable smartcontroller scene executor ...
Wait smartcontroller activation ...
Unlock dropbear service ...
Unlock SSH server ...
Set password "root" for root user ...
Enabling dropbear service ...
Run SSH server on port 22 ...
Test SSH connection to port 22 ...

#### SSH server are activated! ####

==========================================================

Xiaomi MiR Patcher

 1 - Set IP-address (current value: 192.168.31.1)
 2 - Connect to device (install exploit)
 3 - Read full device info
 4 - Create full backup
 5 - Install EN/RU languages
 6 - Install Breed bootloader
 7 - Install firmware (from directory "firmware")
 8 - {{{ Other functions }}}
 9 - [[ Reboot device ]]
 0 - Exit

Select:

```

remittor, 2025-03-06:
Ну тогда в файле "connect5.py" замените строчку
```
hackCheck = gw.detect_hackCheck()
```
на 
```
hackCheck = 2
```

freefelt, 2025-03-06:
К сожалению, ничего не поменялось. Так же сбросил роутер перед этим.

```
==========================================================

Xiaomi MiR Patcher

 1 - Set IP-address (current value: 192.168.31.1)
 2 - Connect to device (install exploit)
 3 - Read full device info
 4 - Create full backup
 5 - Install EN/RU languages
 6 - Install Breed bootloader
 7 - Install firmware (from directory "firmware")
 8 - {{{ Other functions }}}
 9 - [[ Reboot device ]]
 0 - Exit

Select: 2

device_name = RC06
rom_version = 1.1.16 release
mac_address = a8:a1:59:95:84:a7
CountryCode = CN
Enter device WEB password: teanasLife33
WARN: Exploits "arn_switch/start_binding" not working!!!
Detect using xiaoqiang "hackCheck" fix ;-)
Enable smartcontroller scene executor ...
Wait smartcontroller activation ...
Unlock dropbear service ...
Unlock SSH server ...
Set password "root" for root user ...
Enabling dropbear service ...
Run SSH server on port 22 ...
Test SSH connection to port 22 ...

#### SSH server are activated! ####

==========================================================

Xiaomi MiR Patcher

 1 - Set IP-address (current value: 192.168.31.1)
 2 - Connect to device (install exploit)
 3 - Read full device info
 4 - Create full backup
 5 - Install EN/RU languages
 6 - Install Breed bootloader
 7 - Install firmware (from directory "firmware")
 8 - {{{ Other functions }}}
 9 - [[ Reboot device ]]
 0 - Exit

Select:
```

freefelt, 2025-03-06:
Попробовал несколько предыдущих коммитов.
Вот [этот](https://github.com/openwrt-xiaomi/xmir-patcher/commit/fa8dc9d101ec33a217187d1d8939b91fe6e8dd9c) работает

freefelt, 2025-03-06:
Да, вроде все норм так

remittor, 2025-03-06:
А если выполнить
`run.bat connect6.py <WEB_PASSWORD>`
то сразу сломается wifi ?

freefelt, 2025-03-06:
Да, сразу отрубилось и не подключается больше.

```
C:\Users\admn\Downloads\xmir-patcher-main\xmir-patcher-main>run.bat connect6.py teanasLife33
device_name = RC06
rom_version = 1.1.16 release
mac_address = a8:a1:59:95:84:a7
Detect valid SSH server on port 22 (auth OK)
CountryCode = CN
Traceback (most recent call last):
  File "connect6.py", line 63, in <module>
    raise ExploitNotWorked('Exploits arn_switch/start_binding not working!!!')
gateway.ExploitNotWorked: Exploits arn_switch/start_binding not working!!!
```

remittor, 2025-03-06:
Попробуйте самую свежую версию XMiR

freefelt, 2025-03-06:
Да, теперь все хорошо, спасибо!

```
==========================================================

Xiaomi MiR Patcher

 1 - Set IP-address (current value: 192.168.31.1)
 2 - Connect to device (install exploit)
 3 - Read full device info
 4 - Create full backup
 5 - Install EN/RU languages
 6 - Install Breed bootloader
 7 - Install firmware (from directory "firmware")
 8 - {{{ Other functions }}}
 9 - [[ Reboot device ]]
 0 - Exit

Select: 2

device_name = RC06
rom_version = 1.1.16 release
mac_address = a8:a1:59:95:84:a7
CountryCode = CN
Enter device WEB password: teanasLife33
hackCheck version = 2
WARN: Exploits "arn_switch/start_binding" not working!!!
Detect using xiaoqiang "hackCheck" fix ;-)
Enable smartcontroller scene executor ...
Wait smartcontroller activation ...
Unlock dropbear service ...
Unlock SSH server ...
Set password "root" for root user ...
Enabling dropbear service ...
Run SSH server on port 22 ...
Test SSH connection to port 22 ...

#### SSH server are activated! ####

====================================================
```======

closed 2025-03-06 by freefelt