openwrt-xiaomi/xmir-patcher #78: BE5000, 1.0.6 Version

shenohe, 2025-03-08:
[image]

这个版本的不能开启，大佬能兼容处理一下吗？

remittor, 2025-03-08:
Try "install exploit" again after this error.

steven-qi, 2025-04-28:
Just add my 1 cent regarding Xiaomi BE5000 in case it is helpful.

If you see `error_2` when trying to `Install permanent SSH`, it may be caused by mismatched kernel version and the Xiaomi patcher kernel mode (`5.4.0+` vs `5.4.55`).

You can verify this by running `dmesg | grep xiaomi_patcher.ko` (at this stage, you should already have telnet access or even temporary ssh access if you run the script twice), and you should see the following:

```shOut
xiaomi_patcher.ko: version magic '5.4.0+ SMP mod_unload ARMv7 p2v8' should be '5.4.55 SMP mod_unload ARMv7 p2v8'
```

The best solution for this situation might be rebuilt Xiaomi patcher kennel mode with correct kennel libraries, but as the version difference is pretty small, I decided to take a risk to manually patch the version magic:

```sh
sed -i "s/5.4.0+/5.4.55/g" /tmp/xiaomi_patcher.ko
```

then you can try to load the manually:
```sh
/sbin/insmod /tmp/xiaomi_patcher.ko
```

if there is no output, then it means it has been loaded. you can verify it by running `lsmod | grep xiaomi_patcher.ko`

Now, you can try to `Install permanent SSH` again.

(I don't know if the `/tmp/xiaomi_patcher.ko` will be overwritten by `Install permanent SSH`, so I actually executed script I found under `/tmp/ssh_install.sh` folder on the host computer and executed the part after `insmode $KN_mod` manually, and it worked. Then I ran `Install permanent SSH` again with no error.)

remittor, 2025-04-28:
> If you encounter `error_2` when trying to **Install permanent SSH**, it may be caused by a mismatch between the kernel version and the xmir patcher's kernel module (`5.4.0+` vs `5.4.55`).

This is very strange kernel behavior when loading a `kmod` (via `insmod`). 
This shouldn't happen, since the kernel 5.4.x is compiled with these options:
```
CONFIG_MODULES=y
# CONFIG_MODULE_FORCE_LOAD is not set
CONFIG_MODULE_UNLOAD=y
# CONFIG_MODULE_FORCE_UNLOAD is not set
# CONFIG_MODVERSIONS is not set
# CONFIG_MODULE_SRCVERSION_ALL is not set
# CONFIG_MODULE_SIG is not set
# CONFIG_MODULE_COMPRESS is not set
# CONFIG_MODULE_ALLOW_MISSING_NAMESPACE_IMPORTS is not set
# CONFIG_UNUSED_SYMBOLS is not set
# CONFIG_TRIM_UNUSED_KSYMS is not set 
```
And what error does utility `insmod` return?

> However, since the version difference is **relatively small**, I decided to take the **risk of manually patching the version magic**

There is no risk in this action, since I specifically made it so that the `kmod` use only established kernel headers. That is, this module is guaranteed to work on kernels 5.4.x ... 5.14.x
And already with kernel 5.15 there were cardinal changes in the base headers.

steven-qi, 2025-04-28:
> what error does utility insmod return?
There wasn't any useful info from `insmod` other than a general error message.

But `dmesg | grep xmir_patcher.ko` indeed give the following information.

```shOut
xmir_patcher.ko: version magic '5.4.0+ SMP mod_unload ARMv7 p2v8' should be '5.4.55 SMP mod_unload ARMv7 p2v8'
```

remittor, 2025-04-28:
> But `dmesg | grep xmir_patcher.ko` indeed gave the following information, which leaded me to that trick.

Please, test latest commits: https://github.com/openwrt-xiaomi/xmir-patcher/commits/main/

steven-qi, 2025-04-28:
Yes, the latest version works well without version magic error.
However, I still lost SSH access after reboot...

steven-qi, 2025-04-28:
And I tried to execute the following scripts manually from telnet, and it indeed started SSH session:
```sh
/etc/crontabs/dropbearmulti/install.sh
/etc/crontabs/patches/ssh_patch.sh
```

remittor, 2025-04-28:
> I guess that is because that I am running the router in AP mode, thus firewall is disabled automatically.

and I didn't even know about it

steven-qi, 2025-04-28:
I end up with the following solution which works ok so far:

1. create a file `/etc/crontabs/dropbearmulti/maybe_install.sh` with the following content (copied from `install.sh` with most content removed)

  ```sh
  #!/bin/sh

  XDIR=/etc/crontabs/dropbearmulti

  if [ ! -d /etc/dropbear ]; then
    $XDIR/install.sh
  fi
  ```

2. make it executable by

  ```sh
  chmod a+x /etc/crontabs/dropbearmulti/maybe_install.sh
  ```

3. add a cron job by running `crontab -e` with the following

  ```
  * * * * * /etc/crontabs/dropbearmulti/maybe_install.sh && /etc/crontabs/patches/ssh_patch.sh
  ```