openwrt-xiaomi/xmir-patcher #79: Install firmware RD12 in RD04 error

JuanManuelRomeroGarcia, 2025-03-18:
device: "RD04"
img_write = True
Image files in directory "firmware/":
  "firmware/miwifi_rd12_firmware_5724d_1.0.75_INT.bin"
Download file: "/tmp/dmesg.log" ....
Download file: "/tmp/mtd_list.txt" ....
Download file: "/tmp/mtd_info.txt" ....
Download file: "/tmp/mtd_fdt.txt" ....
Download file: "/tmp/kcmdline.log" ....
Parse all images...
Stock image list: {'xiaoqiang_version': 531, 'firmware.bin': 14476514}
Parse HDR image for "RD12" router
Stock image list: {}

ERROR: Stock image is empty!

remittor, 2025-03-18:
> Parse HDR image for "RD12" router

firmware images "cs6c" not supported

JuanManuelRomeroGarcia, 2025-03-19:
device: "RA70"
img_write = True
Image files in directory "firmware/":
  "firmware/miwifi_ra70_firmware_a79cd_3.0.40_INT.bin"
Download file: "/tmp/dmesg.log" ....
Download file: "/tmp/mtd_list.txt" ....
Download file: "/tmp/mtd_info.txt" ....
Download file: "/tmp/mtd_fdt.txt" ....
Download file: "/tmp/kcmdline.log" ....
Parse all images...
 
ERROR: HDR2 stock image not supported!