openwrt-xiaomi/xmir-patcher #80: install FIrmware

JuanManuelRomeroGarcia, 2025-03-19:
/home/XiaoHack/xmir-patcher-main/xmir_base/fdt/misc.py:67: SyntaxWarning: invalid escape sequence '\*'
  text = re.sub('//.*?(\r\n?|\n)|/\*.*?\*/', '\n', text, flags=re.S)
Detect valid SSH server on port 22 (auth OK)
device: "RA70"
img_write = True
Image files in directory "firmware/":
  "firmware/miwifi_ra70_firmware_a79cd_3.0.40_INT.bin"
Download file: "/tmp/dmesg.log" ....
Download file: "/tmp/mtd_list.txt" ....
Download file: "/tmp/mtd_info.txt" ....
Download file: "/tmp/mtd_fdt.txt" ....
Download file: "/tmp/kcmdline.log" ....
Download file: "/tmp/kver.txt" ....
Parse all images...
 
ERROR: HDR2 stock image not supported!

remittor, 2025-03-20:
> /home/XiaoHack/xmir-patcher-main/xmir_base/fdt/misc.py:67: SyntaxWarning: invalid escape sequence '*'
> text = re.sub('//._?(\r\n?|\n)|/*._?*/', '\n', text, flags=re.S)

Fixed https://github.com/openwrt-xiaomi/xmir-patcher/commit/6cc2a127497d3c5a9ea039addbd10e904a89d319

remittor, 2025-03-20:
> ERROR: HDR2 stock image not supported!

Sorry

closed 2025-10-19 by remittor