openwrt-xiaomi/xmir-patcher #81: CB0401V2 Warnings

albatrosify, 2025-03-25:
device_name = CB0401V2
rom_version = 3.5.1 release
mac_address = 88:d8:2e:e5:6e:e2
CountryCode = DE
hackCheck version = 3
WARN: Exploits "arn_switch/start_binding" not working!!!
WARN: Exploits "Smartcontroller" are not usable (hackCheck:3)

remittor, 2025-03-25:
> rom_version = 3.5.1 release

Did the device update itself to this firmware?

albatrosify, 2025-03-25:
I received the device like this from the shop

albatrosify, 2025-03-26:
Ok, thanks for your help. I will try to find a link to the firmware 3.5.9 with the support. 
From where do you have the information, that 3.5.9. is available for the 5G CPE PRO V2?

I dont know yet what benefit the Exploit will bring me, but my hope is that I can change the Router Advertisement and set up a custom DNS Server which is running locally here.
I do have an adguard adblock dns server running in a proxmox lxc, and would like to set it up in the XIAOMI either as its primary DNS or as DHCPv4 DNS for the clients and via RA then...

davidohne, 2025-05-15:
> Ok, thanks for your help. I will try to find a link to the firmware 3.5.9 with the support.

Did you have any success receiving the update file? Do you have a provider branded version (Telekom/Magenta is selling these devices a lot) or do you have a Xiaomi/Unbranded one?

svenvogel, 2025-05-15:
I have the same question.