openwrt-xiaomi/xmir-patcher #88: RD04v2 Warnings

mandiyama, 2025-04-12:
device_name = RD04V2
rom_version = 1.0.29 release
mac_address = censored
CountryCode = CN
hackCheck version = 3
WARN: Exploits "arn_switch/start_binding" not working!!!
WARN: Exploits "Smartcontroller" are not usable (hackCheck:3)

remittor, 2025-04-21:
Flash old firmware:
https://cdn.cnbj1.fds.api.mi-img.com/xiaoqiang/rom/rd04v2/miwifi_rd04v2_firmware_51750_1.0.16.bin

mandiyama, 2025-04-21:
When installing normally, it is not possible to install the update, I guess if I try to install via **MIWIFIRepairTool** or are there other alternatives to install the 1.0.16 update?
I also tried to install it from computer, but without success 

[image]

remittor, 2025-04-22:
> if I try to install via **MIWIFIRepairTool**

This is exactly the method needed.

caizhenbing, 2025-07-06:
https://bigota.miwifi.com/xiaoqiang/tools/MIWIFIRepairTool.x86.zip

closed 2025-08-27 by mandiyama