openwrt-xiaomi/xmir-patcher #93: RN06 not yet worked.

JuanManuelRomeroGarcia, 2025-04-21:
device_name = RN06
rom_version = 2.0.60 release
mac_address = d4:5d:64:7f:e0:56
CountryCode = ES
Enter device WEB password: a5057014
hackCheck version = 3
WARN: Exploits "arn_switch/start_binding/set_mac_filter" not working!!!
WARN: Exploits "Smartcontroller" are not usable (hackCheck:3)

Good job with the BE5000 with my friend Fran

remittor, 2025-04-21:
Needed downgrade to https://cdn.cnbj1.fds.api.mi-img.com/xiaoqiang/rom/rn06/miwifi_rn06_firmware_60fc5_2.0.40_INT.bin

JuanManuelRomeroGarcia, 2025-04-21:
[image]

JuanManuelRomeroGarcia, 2025-04-21:
[image]