openwrt-xiaomi/xmir-patcher #96: BE3600  Connect to device (install exploit)

freyya-zero, 2025-05-12:
device_name = RD15
rom_version = 1.0.87 release
mac_address = 
CountryCode = CN
hackCheck version = 3
WARN: Exploits "arn_switch/start_binding/set_mac_filter" not working!!!
WARN: Exploits "Smartcontroller" are not usable (hackCheck:3)

remittor, 2025-05-12:
Needed downgrade firmware via bootloader

WangN2000, 2025-05-14:
小米刷砖工具降不了，有没有其他工具可以降级

jackathere, 2025-05-23:
> device_name = RD15 rom_version = 1.0.87 release mac_address = CountryCode = CN hackCheck version = 3 WARN: Exploits "arn_switch/start_binding/set_mac_filter" not working!!! WARN: Exploits "Smartcontroller" are not usable (hackCheck:3)

> 小米刷砖工具降不了，有没有其他工具可以降级

"I’m having the same issue. I tried downgrading the firmware using the Xiaomi Router Repair Tool, but it didn’t work either."

ErikNCB, 2025-07-30:
Try setting a TFTP server with the firmware file and static ip to 192.168.31.100, then set the router to TFTP recovery mode starting it while pressing the reset button for 1 seconds

I have not tried yet, but thats the instructions given by chatgpt

dolbinski, 2025-10-31:
Not tested yet as I received the router (be5000) yesterday . It says should work for be3600 as well. (Deepseek)
1. Prepare for Recovery Mode:
   · Configure your computer's IP address manually.
   · Go to your computer's Network Settings -> Ethernet Adapter Properties -> Internet Protocol Version 4 (TCP/IPv4).
   · Select "Use the following IP address":
     · IP address: 192.168.31.2
     · Subnet mask: 255.255.255.0
     · Default gateway: 192.168.31.1
   · Leave DNS blank.
2. Enter Recovery Mode:
   · Power off the router.
   · Hold the reset button on the back of the router (you may need a pin). Keep holding it.
   · Power on the router while still holding the reset button.
   · Continue holding the reset button for about 5-10 seconds until the Yellow/Amber indicator light begins to blink slowly. Then, you can release the button.
   · The router is now in Recovery Mode.
3. Access the Recovery Interface:
   · On your computer, open a web browser and go to http://192.168.31.1.
4. Flash the Firmware in Recovery:
   · You should see a very simple, minimalist page (often blue and white) with a button to "Choose File" or "Upload Firmware".
   · Select the older .bin firmware file and click "Upload".
   · The flashing process will begin. Again, do not interrupt this process. It may take longer than a normal update.
5. Reboot and Revert Network Settings:
   · Once complete, the router will reboot. Wait for it to fully start up.
   · Go back to your computer's network settings and change the IP configuration back to "Obtain an IP address automatically" (DHCP).