openwrt-xiaomi/xmir-patcher #119: AX3000T V3 2.0.12

hhxxc, 2025-09-29:
hackCheck version = 3
WARN: Exploits "arn_switch/start_binding/set_mac_filter" not working!!!
WARN: Exploits "Smartcontroller" are not usable (hackCheck:3)
WARN: Exploit "get_icon" not working!!! (API not founded)

helphelphelphelphelphelphelphelphelp

Misfireee, 2025-11-16:
same helppp

niekvugteveen, 2025-11-26:
It executes the wrong connection method (connect5.py).

First of all:
- Resetbutton to restore factory settings (30 seconds)
- go to 192.168.31.1
- follow wizzard
- continue setup without connecting a cable
- workmode: router, not AP
- choose DHCP
- UNCHECK make wifi password same as admin password. Both passwords can be identical.

Run connect4.py first. I will activate MESH (netmode:4). First time it will likely fail (well it will succeed and throw an exception technically) and disconnect. But once it has been restarted you can run it again. Like so:

.\run.bat connect4.py
device_name = RB01
rom_version = 1.0.83 release
mac address = xxxxxxxxx
Current CountryCode = NL
Enter device WEB password: yourpassword
netmode = 0. Attempting to set netmode to 4 ...
Forced "whc_cap" mode enabled! (netmode=4)
Traceback (most recent call last):
  File "d:\python\python38\Lib\site-packages\urllib3/connection.py", line 174, in _new_conn
  File "d:\python\python38\Lib\site-packages\urllib3/util/connection.py", line 96, in create_connection
  File "d:\python\python38\Lib\site-packages\urllib3/util/connection.py", line 86, in create_connection
socket.timeout: timed out

Wait for router to reboot.

Run it again
.\run.bat connect4.py
device_name = RB01
rom_version = 1.0.83 release
mac address = xxxxxxx
Current CountryCode = NL
Enter device WEB password: yourpassword
Run SSH server on port 22 ...

#### SSH and Telnet services are activated! ####

you can now proceed with option 7 in the menu (install firmware), make sure you downloaded and put the bin file in the firmware dir.

ahmedtalaat327, 2025-11-29:
> It executes the wrong connection method (connect5.py).
>
> First of all:
>
> * Resetbutton to restore factory settings (30 seconds)
> * go to 192.168.31.1
> * follow wizzard
> * continue setup without [...]

are you sure this works for RD03v2 ? opening the sSh port?

EgorBlyablin, 2025-11-30:
> No it works on RD01. Read rest of the thread RD03 is not supported yet

Title strictly says: problem with **RD03v2 2.0.12** (that's latest firmware). Other revisions are unrelated to the current issue.

Have anyone tested such SSH activation method on RD03v2? If it doesn't work, it shouldn't be here at all

JimmyMoriarty, 2026-01-03:
Such a waste for deive with 256 RAM and 125 ROM. Xiaomi ruining the fun

iboros445, 2026-02-18:
> https://github.com/openwrt-xiaomi/xmir-patcher/blob/04a789e0ef62f1f3515805ad2843bf7c22f51150/xqmodel.py#L121
> Support seems to be added already.

Has anyone had a positive experience with it and installing OpenWrt on a Qualcomm model?

wang-zhix, 2026-02-26:
My version is 2.0.12, I didn't upgrade to 2.0.28, but it still shows an error
WARN: Exploits "arn_switch/start_binding/set_mac_filter" not working!!!
WARN: Exploits "Smartcontroller" are not usable (hackCheck:3)
WARN: Exploit "get_icon" not working!!! (API not founded)

ADCDS, 2026-09-28:
Update: found a way to install without UART. Writeup: https://blog.adriel.eu/xiaomi-cabmeshd/
Installer + guide: https://github.com/ADCDS/xmir-patcher/blob/connect8-connect9/RD03v2-INSTALL.md